prusa3d/PrusaSlicer #11566: Preferences missing a scrollbar

JimF42, 2023-11-01:
### Description of the bug

Hello,

I am running PrusaSlicer 2.6.1 on a laptop. The laptop runs natively at 1920x1080 and at a scale of 150%.
[image]

When I open the Preferences screen, the very last item is cut off. I only realized it existed due to a visual artifact, the top of the checkbox.
[image]

The missing item, in my case appears to be `Enable support for legacy 3DConnexion devices:`
[image]

Please consider adding a scrollbar to this, and any window that may be too long. You do this already with the tabs like Print Settings and Printer Settings.

### Project file & How to reproduce

This issue is not project specific, but I have included an empty project so you have the config values. I simply renamed it from .3mf to .zip since it is inherently a zip file.
[Empty.zip](https://github.com/prusa3d/PrusaSlicer/files/13226375/Empty.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1+win64

### Operating system

Windows 11 Pro 22H2

### Printer model

MK4

Jan-Soustruznik, 2023-11-01:
Hello, @JimF42 , 

thank you for your report. Could you check with our latest released alpha1, please?  (link to changelog: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). We did there some improve of user interface.

Thank you, have a nice day.

JimF42, 2023-11-01:
Hello @Jan-Soustruznik,

Wow, that was a quick fix! 🤣

Yes, it is in 2.7.0-alpha1. The only issue I had, and it's minor and self-resolvable, is that the scrollbar visibility isn't taken into account when sizing the window. Of course, I can resize the window now, so I can fix that--but, it's just fixed once. If I quit 2.7a1 and later restart it, the window is cut off again.

I'm sure this will be fixed in Alpha2!!
[image]

Jan-Soustruznik, 2023-11-02:
SPE-2010

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena