prusa3d/PrusaSlicer #11567: Support Catch2 v3

onitake, 2023-11-01:
**Is your feature request related to a problem? Please describe.**
The PrusaSlicer package on Debian is built with the OS-provided Catch2 testing framework instead of the vendored version in the Slicer source code.
This package is currently being updated to the next major upstream version, which has several incompatible changes.

**Describe the solution you'd like**
Slicer should upgrade from Catch2 v2 to v3.
Additionally, it would be very helpful if it didn't vendor Catch2, or at least support building against a preinstalled version if available.

**Describe how it would work**
Upgrading to Catch2 v3 will require several changes to the unit tests, to account for incompatible upstream changes: https://github.com/catchorg/Catch2/blob/devel/docs/migrate-v2-to-v3.md

**Describe alternatives you've considered**
I've started preparing a custom patch for the Debian version of Slicer, but it would be better if the upstream sources are upgraded instead.

**Additional context**
Debian bug: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1054697

Upgrading to Catch2 v3 would probably also solve issue #5400 .

tamasmeszaros, 2023-11-02:
Hi! The next upcoming version (2.7-alpha2) of PrusaSlicer will be able to handle external catch2 library, but only the 2.x version line. Unfortunately, the tests were written for version 2.x, and I'm pessimistic about any time estimate about adjusting all the tests to be compatible with 3.x.
If you have some patches that does the job, we would be grateful for a PR.

onitake, 2023-11-02:
I actually have a patch for you, but it's only been tested with the Debian build process and may need tweaking.
I'd be happy to push a preliminary PR, though.

stixpjr, 2024-04-14:
I support a port of prusaslicer in NetBSD pkgsrc (https://github.com/NetBSD/pkgsrc-wip/tree/master/prusaslicer/), and have run into the same issue, with pkgsrc-2024Q1 upgrading to Catch2 v3.5.3. For now, I'll see if I can patch this PR into the build.

renatoaguiar, 2024-04-14:
> I support a port of prusaslicer in NetBSD pkgsrc (https://github.com/NetBSD/pkgsrc-wip/tree/master/prusaslicer/), and have run into the same issue, with pkgsrc-2024Q1 upgrading to Catch2 v3.5.3. For now, I'll see if I can patch this PR into the build.

I had the same issue with the OpenBSD port (https://github.com/openbsd/ports/tree/master/cad/prusaslicer). I hope this gets merged, so I can get rid of several patches :)

onitake, 2026-09-13:
Looks like this was fixed with commit 2b01d14f7b7389ba313a083e8ca07b2681f81543 .

closed 2026-09-13 by onitake