prusa3d/PrusaSlicer #11569: 2.7.0-alpha1 generates invalid gcodes

sammcj, 2023-11-02:
### Description of the bug

Testing out PrusaSlicer 2.7.0 Alpha 1 and I found it to generate invalid gcodes.

No config changes have been made from 2.6.x, the same project is loaded in, gcode is generated and Klipper shows it to be invalid.

Gcode and Project attached.

### Project file & How to reproduce

- [foot-for-foot_2h26m_0.24mm_234C_PETG_ENDER5S1.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/13233617/foot-for-foot_2h26m_0.24mm_234C_PETG_ENDER5S1.gcode.zip)
- [foot-for-foot.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13233618/foot-for-foot.3mf.zip)

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0-alpha1+MacOS-arm64

### Operating system

macOS 14.1

### Printer model

Ender 5 S1

rtyr, 2023-11-02:
Thanks, we will fix it for next release. It happens with combination of Klipper g-code flavor and label objects set to "octoprint-comments" (it was the only option in older PS versions).

It should work fine if you switch to "firmware-specific"

[image]

lukasmatena, 2023-11-02:
SPE-2006

sammcj, 2023-11-02:
Ah, glad you knew about this one! Thanks and keep up the amazing work - you're all appreciated greatly.

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena