prusa3d/PrusaSlicer #11570: Strange extrusion issue on part in multiple layers.

BrienKing, 2023-11-02:
### Description of the bug

The problem I get is a small over extrusion on a layer or two.  It's not a layer shift because the extra extrusion is all the way around the part evenly.

I expected the walls to be even all the way up the part.

It might be hard to see in the picture, but it's very visible in person.
[image]

[image]

If you go into the slicer and turn the legend from "Feature" to "Layer time (linear)", it shows where the line appears.  The light green line in the picture below appears to be the issue.

[image]

I've attached a 3mf file with the part.
[Front Cover.zip](https://github.com/prusa3d/PrusaSlicer/files/13235905/Front.Cover.zip)

### Project file & How to reproduce

Project file is attached.   

Print it.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1

### Operating system

Windows 10

### Printer model

Prusa XL Dual Extruder (alpha 2)

rtyr, 2023-11-02:
This is similar to https://github.com/prusa3d/PrusaSlicer/issues/991 and other issues. It is unfortunaly very hard to compensate for various physical influences (filament material thermal expansion coefficient, cooling, print enviroment, filament properties - dryness, composition)

See:
https://github.com/prusa3d/PrusaSlicer/issues/991#issuecomment-458614931

https://help.prusa3d.com/article/the-benchy-hull-line_124745

I am closing this as a duplicate. Your issue will be linked.

rtyr, 2023-11-02:
As a side note, the issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3412 is not related to this (it is known issue with 5.1.0-alpha firmware).

closed 2023-11-02 by rtyr