prusa3d/PrusaSlicer #11572: AI Generated Supports

LuckyHeroes, 2023-11-02:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Possible to train AI model to generate supports that are the best suited for a particular design with a selection of options to customise the supports for your needs.

**Describe how it would work**
Add another option for supports "AI Support", a user would select this option and a trained AI model would analyse the 3d model for the "BEST" solution for supports, this would have 3 options for the user (least material/best quality/balanced). 

**Describe alternatives you've considered**
None

**Additional context**

https://youtu.be/kSV2JeXQgm4?si=6_TDqmf9L6Py0PLY

Youtube - Maker's Muse

PaparazziN, 2023-11-15:
Hi @LuckyHeroes!
Whilst I agree that the idea of AI generated sounds interesting, I think the automatic painting feature introduced with PS2.6 is a pretty close match to what you are describing functionally. From my experience it has been a very seamless experience which provides enough support for all prints. It is currently a bit limited in what you can change but adding more customization as raised in #9806 would definitely be an improvement. 
If you want to keep this request open I would suggest that you specify what the AI would do different than the current implementation of the automatic painting tool.

bryn51, 2024-07-20:
Ok, rising to the challenge in the previous post:
Currently prusa slicer (2.8.0) makes a reasonable job of supports but a couple of key areas could be improved.
1. Supports beneath horizontal surfaces: the model surface being supported is not of great quality and looks very much unlike the surface of vertical and upwards horizontal surfaces.
Improvement sought: Improve the surface quality of supported surfaces.
2. Adherence of same material supports is sometimes excessive and its related to the material used. TPU is an example, where attempts to remove the supports leave a model surface that is severely degraded. This is made worse by the fact that regular MMU systems will not handle FLEX very well, making it difficult to apply dissimilar support material. 
The solution would necessarily include special attention to this issue to see if a particular interface design can be developed better suited to FLEX and other materials (PP) which adhere too strongly to itself for regular supports to work as intended.

It may well be the case that existing PS features could be used towards achieving these goals, but requiring expertise to set them exactly correct. Which is where an AI might come in: to have the Slicer detect that the special attention is required and making tweaks to several settings to achieve the best outcome.

Dimensional adjustment is another area where an AI in PS could help (off topic for this request)

closed 2026-09-29 by prusaslicer-bot[bot]