prusa3d/PrusaSlicer #11574: 2.7.0-a1 Temperature Variation Input

BaconFase, 2023-11-02:
### Description of the bug

The Temperature Variation setting is clearly for zero to negative values but the input box won't allow for the keyboard minus sign as an input. So you'd have to click and hold the down arrow on keyboard/GUI to actually set a correct temperature delta.

Please allow for the minus sign input.

### Project file & How to reproduce

Print Settings > Multiple Extruders > Ooze Prevension > Temperature Variation

Try to input minus sign (-) next to number row or numpad. PS will not allow and computer will chime.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0-a1

### Operating system

Win10

### Printer model

XL-5T

YuSanka, 2023-11-03:
SPE-2015

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena