prusa3d/PrusaSlicer #11578: Enable use of use interface shells on intersecting models

murk-sy, 2023-11-02:
Currently, when printing different materials, depending on the combination, having fairly dense interfaces may be required. However, modelling those can be quite difficult and resource consuming in 3D modelling software.

### Proposed solution
When having a model with multiple parts loaded, if there are intersecting parts of the models, the intersect of those two models should be considered the "interface volume".

**Side view** example: yellow is the intersection of the 2 models - an example would be the solid and flexible parts of the tyre.
[image]

With slicing, the material which is used for the intersection alternates:
[image]

I believe this should be *relatively* simple to implement compared to some sort of interface painting or something like that.
If the intersection isn't chamfered and met on point, it could lead to situations where for example a wheel would have a different inside (larger rim) than outside when even number of layers is used. I believe this wouldn't be a major issue - or, in case the adhesion between materials isn't perfect, having multiple solid layers near the outside is likely better, but that's something that can easily be done during modelling.

### Stronger interfaces
Currently, the interface consists only of entire layers, which will likely be perfectly functional for most cases.

However, for materials that *really* don't like each other, some form of hooking may be required.

**Top down view** example:
Odd layer:
[image]
Even layer:
[image]

The sketch is a bit terrible, but hopefully it should be a decent enough indication of what I have in my mind. I'll make an actual model or printable example if I'll have time in the coming days.

By using sort of a hook system that is automatically being offset, blue is printed over orange, but the corners of blue (or ornange) are always connecting to themselves.

Therefore, even if 2 materials have terrible adhesion, this sort of interface should be stronger than probably just about anything you could do with injection molding.

freeformz, 2024-08-13:
I am running into this issue on my XL where 2 materials are printed "against" one another, but not really locked together.

murk-sy, 2025-02-18:
Implemented in 2.9.1-alpha1, closing

closed 2025-02-18 by murk-sy