prusa3d/PrusaSlicer #11583: Creality K1 max support - Multiple fans

RubenMeneses, 2023-11-03:
Hi,
From what I have been able to find while searching, there is no support for the creality k1 printers.
I have been trying to reverse what creality print does with the 3 fans it has.
If I wanted to try and contribute to the project, with the way that the how PrusaSlicer is structured, how could try and contribute?

rtyr, 2023-11-07:
Here is some info about 3rd party profiles https://github.com/prusa3d/PrusaSlicer/wiki/Vendor-bundles-and-updating-process.

Creality bundle was maintained mainly by @pmjdebruijn so far.

Kaelum, 2023-12-24:
@rtyr in order to work with the many printers that currently exist, and the ones still coming, PrusaSlicer requires the ability to control multiple fans.  At the very least it needs to be able to control:

1. Part cooling fan. (exists)
2. Auxiliary part cooling fan.
3. Exhaust fan.
4. Chamber heater / stirring fan.

This is the bare minimum number of fans that need to be controllable.  Fan # 2 may need layer control, with a static or ramp up setting.  Fans # 3 and # 4 just need a static setting.  The number of fans, alias names, and type of control for the fans, should be defined in the Printer Setting.

Kaelum, 2023-12-24:
@RubenMeneses & @pmjdebruijn if you need something temporary for now, you can add this to your `Printer Settings / Custom G-code / After layer change G-code` section:

```
{if filament_type[0] =~ /PLA/ && layer_num > disable_fan_first_layers[0]}
{if layer_num >= full_fan_speed_layer[0]}M106 P2 S128;{endif}
{endif}
```

It will set the auxiliary part cooling fan speed to 50% (128) at the `Full fan speed layer` setting that you set on the filament.  For a different speed, you will need to change the custom G-code to a different value.

Thovthe, 2026-01-13:
@rtyr @pmjdebruijn @RubenMeneses 
This should be **CLOSED** as https://github.com/prusa3d/PrusaSlicer-settings/pull/244 was merged into the settings repository.

closed 2026-09-29 by prusaslicer-bot[bot]