prusa3d/PrusaSlicer #11586: 2.7.0 alpha win64 profiles lost

Dylena, 2023-11-04:
### Description of the bug

I downloaded and put the folder in programfiles with a shortcut to my desktop, running next to an older 2.6 x because of my Elegoo Neptune 4 Pro not having a real profile yet and afraid updating will lose the profile.

And... that is exactly what happened when I wanted to go print today after putting a lot of effort completing my Elegoo Neptune 3 Max profile in 2.7.0: all gone only default left

Thanks for the program.
Kind regards.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1+win64

### Operating system

Windows10

### Printer model

Elegoo Neptune 3 Max

rtyr, 2023-11-04:
Alpha releases are using different data directory, so you will not see those profiles from 2.6.1. You can start that older 2.6.x, export profiles to the configuration bundle, then start 2.7.0-alpha and import it there.

Dylena, 2023-11-08:
But I used Alpha 2.7.0 for another printer that wasn't configured in 2.6.1
So, I don't have a backup of what I did. Just reporting the loss, maybe it
happened before.

I went back to install the new 2.6.x in time when I have a real profile for
my Neptune 4 Pro from Elegoo then I can bring them all together. Just hope
they don't bite another, that is why 2 slicers for now.

rtyr, 2023-11-08:
Your profiles may be not visible for various reasons (I mean expected reasons). For example because of compatible condition or if your printer profile is from different vendor space than your previously created profile.

If you send us zip archive with your configuration folder, we can check it (help -> show configuration folder].

Dylena, 2023-11-08:
Thanks again for the reply,

aww, I immediately thought, I don't want this to happen anymore so I
deleted the 2.7.0
I am very new but learning fast in Prusa from CURA. I saved all those 3
tabs on their own and I thought like in 2.6 it would stay.

But when I came up the next day all that work I put in to configure was
gone. There was only this 'default' word I could choose from.
Thanks for the reply. In the first place I merely sent it to add to the
statistics. If other people had this as well it might be worth a look.
I wrote all the conditions where it happened. We'll see if it matters or if
I did something wrong.

I went back to CURA for the Elegoo Neptune 3 Max for now but hope to have
some time later to configure this 2.6.x latest and test prints with that.
Also finally dared to update my 2..6.1 to the +win64 version latest and the
profiles stayed.
Last print was 26 hours for a scale 1:3 toolset with PRUSA and some guessed
/ adjusted numbers partially from the experimental Elegoo Neptune 4 Pro
template.

Some minor things it seemed when printing upwards, not sure what it can be;
hardware, software or both.
Kind regards

[image: Tool set FDM gereedschap (1).jpg]

rtyr, 2023-11-23:
OK, I am closing the issue.

closed 2023-11-23 by rtyr

just-trey, 2023-11-26:
> e configuration bundle, then start 2.7.0

@rtyr what is the new location, please?

Dylena, 2023-11-27:
Have been waiting for 2.6.1 to fix itself but it didn't either. It kept trying to update, no matter if I did modify or repair it would upgrade and in the very end said something was wrong.

With the 2.7 going final now it seems to finally gone and did remember my profiles