prusa3d/PrusaSlicer #11588: Multi material interlocking for multiple objects.

Pixelplanet, 2023-11-04:
With Prusaslicer 2.6.1 the feature of interlocking layers of **paint on multi color areas** was added and its a nice start.
but building on top of this it would be nice if it would be possible to interlock multiple objects.

right now as far as i know you can only have one object and paint on it and the painted on part can interlock with the rest, my proposal would be to give us an option to either tell the slicer to interlock object on the internal layers automatically so from the outside you couldnt even tell the difference between interlocked parts and those that have not been interlocked internally and/or extend the functionality of the paint tool so you can paint on where two objects are supposed to interlock.

ideally both options should be done at some point, automatic interlocking and paint on interlocking for cases where the automatic one doesnt provide the desired result.

skenecroft, 2023-12-04:
I'd love to see this feature added

Pixelplanet, 2023-12-20:
could someone label this as feature request?

cakesmith, 2023-12-21:
+1 for this feature, I think it's great for painted multi-material, but I have some files that are separate pieces, and I would like them to benefit from this as well.

Pixelplanet, 2024-07-24:
Can anyone lable this as a feature request please?

Cinderhaze, 2024-07-28:
Saw someone request something matching this description the other day on reddit ( (wanting to 'sandwich overlapping materials for testing strength) - https://www.reddit.com/r/prusa3d/comments/1ec3e6a/comment/lexfa7g/ - I would love to see an option for layer-overlapping objects..  I was trying to do a quick test print with my MMU3 and placed 3 overlapping boxes hoping to do something like this, and just got a slicer warning about overlapping toolpaths... I know that I can just 'add part' and/or 'add modifier' to put them together and paint it, but I still ended up lacking a good overlap (and the perimeter walls next to each other of the different colors/filaments came apart easily).  It would also be good to do something like an internal 'dovetail' join on the layers for the overlap, or something like that for additional increased strength for differing materials.... i.e. if I wanted to use a TPU/flexible joint within a standard PLA/PETG print.

Pixelplanet, 2024-12-16:
this requests are over a year old now and still not properly labeled, who can be contacted to get it labeled?

lukasmatena, 2025-02-18:
Interlocking originally developed in Cura was implemented in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-18 by lukasmatena