prusa3d/PrusaSlicer #11590: Backdrop option for text modifiers or possibly for all of them

Pixelplanet, 2023-11-04:
Specifically when printing text into an object with a filament thats partially translucent or even an opaque one printed very thin you can often see the other filaments shine through or at least impact the color of the text.

This can probably be seen the best if you try to add text to a black part as the black will cause any lighter color to look dark and "dirty"
for this reason i would like an option to add a backdrop layer to text or possibly to all modifiers you can do so you can use another extruder to print a white layer or what ever else you want behind your text to make it come out exactly as you want.

heres an example how big the difference can be all prints are a single layer at 0.32mm the ones in the back are a random white, Prusa sliver and a random black filament, the ones in the front are Prusament Orange PETG on top, a random opaque light blue in the middle and Prusament Azure blue PLA in the bottom.

[image]

as you can see the background color makes a huge difference to the appearance of the color, this effect is even stronger on lighter colors like pink, yellow or any very bright and shiny color.

right now my workaround is to add a modifier cube resized to be a thin piece that i carefully move behind the text so i can print some different layers behind the text but overall it would be a lot easier if we had a backdrop option to either place it only behind the text itself or behind the entire area the text is occupying.

Pixelplanet, 2023-12-20:
could someone label this as feature request?

Pixelplanet, 2024-07-24:
Can anyone lable this as a feature request please?

Pixelplanet, 2024-12-16:
this requests are over a year old now and still not properly labeled, who can be contacted to get it labeled?

costachedaniel, 2025-01-12:
Resubmit the request. You can now label them yourself.

Pixelplanet, 2026-09-21:
this is still a feature request that has not been labeled and still needs to be implemented.