prusa3d/PrusaSlicer #11596: Respect negative Volumes on "Out of Print Area" check

gaiuscosades, 2023-11-05:
### Description of the bug

When placing objects that exceed the print area slicing cannot be started even if a negative volume does limit the object to be printed to the plate area.

I encounter this frequently when tuning tolerences on parts of objects where I instanciate the object multiple times and only want to print a small part of the object while cutting away the rest using negative volumes.

### Project file & How to reproduce

1. Instantiate an object that exceeds the print area.
2. Add negative volumes that cover the whole object outside the available print volume
3. Press "Slice now".
[out_of_bed_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/13259032/out_of_bed_bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1+win64

### Operating system

Windows 11 Home Build 22621

### Printer model

Prusa Mini+

nharrer, 2025-02-10:
This bug still exists. If anyone encounters this issue, a workaround is to trim the sides using the cut feature, even if there's nothing to print due to the negative volume. Unfortunately, this irreversibly modifies the model. So it would still be great if this could be fixed.

Chose99, 2025-03-14:
I have the same issue and i can cut my part ( as im replacing it from time to time with a simular stl and the negative volums need to stay in place relativly)
Luckly i can get everthing to fit if i arrange manuely.

GudjonIV, 2026-08-25:
A workaround that I've used when encountering this issue is to add the negative volumes so it fits and then right clicking the model and using the `Export as STL/OBJ..` to export it and then import it again which causes the negative volume to become permanent so it fits.

puterboy, 2026-09-02:
Keep this alive until fixed!