prusa3d/PrusaSlicer #11597: Emit temperature commands automatically: support Klipper PRINT_START

viesturz, 2023-11-05:
### Description of the bug

On Klipper it's typical to pass initial temps to the PRINT_START macro and let printer smartly prepare for printing, including any thermal soaking, etc.

At some point PrusaSlicer has started forcing preheat in the GCode with [some heuristic](PRINT_START TOOL_TEMP=[first_layer_temperature[initial_tool]] BED_TEMP=[first_layer_bed_temperature] TOOL=[initial_tool]) to detect when that is not needed.

I woudl recommend to check if the script reads the temp variables instead of checking for the commands.
That would be more robust across different firmwares and Gcode flavors.

https://github.com/prusa3d/PrusaSlicer/blob/e86cd23b452f43972d3cbb53abe97ce5e7fe0cec/src/libslic3r/GCode.cpp#L1716

### Project file & How to reproduce

Use the following start GCode:
PRINT_START TOOL_TEMP=[first_layer_temperature[initial_tool]] BED_TEMP=[first_layer_bed_temperature] TOOL=[initial_tool]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Win11

### Printer model

voron 2.4

lukasmatena, 2023-11-06:
Could you please expand on what problem are you trying to solve? I believe that all you need to do is to disable the checkbox "Emit temperature command automatically" (which was added specifically for this use case, see #2420) and all should be great.

viesturz, 2023-11-06:
Yes disabling the checkbox is a valid workaround.

But I spent 1 hour debugging what the hell was wrong after setting up a new
Voron profile.
I imagine I'm not alone here. And since the automation exists and is
enabled by default, it should work by default.

Another possible fix would be to disable it in Klipper based profiles.

rtyr, 2023-11-07:
> On Klipper it's typical to pass initial temps to the PRINT_START macro

As far as I know, it is mainly typical for Voron users, but its not needed at all for running Klipper printer. 

I really don't see any issue here. You can just disable the checkbox and set what do you want.

rtyr, 2023-11-14:
> In that case please change the default for Voron profiles.

Disabled with configuration update 1.0.4. Closing.

closed 2023-11-14 by rtyr

marcomania75, 2023-12-04:
i am happy I found this, suddenly my little 0.1 wasn't working anymore,... canceled every new print...

ogabriel, 2024-03-20:
> Yes disabling the checkbox is a valid workaround. But I spent 1 hour debugging what the hell was wrong after setting up a new Voron profile. I imagine I'm not alone here. And since the automation [...]

I also agree this should be turned off by default on klipper profile

I also spent some minutes trying to figure out why START_PRINT macro started to suddenly heat up  on the start LoL