prusa3d/PrusaSlicer #11600: Prusa XL Multi Tool

sith77, 2023-11-05:
when using Prusa Slicerwith a multitool XL unit, if you want to change the color in one of the extruders during mid Print there seems to be a missing option. With the option for change filament on the menus, it gives you no option to select which extruder you want to change the filament on . It only defaults to the extruder that’s currently printing, but if you’re using two separate colors on a print, and the colors are intertwined in the object, you should be able to select one of the extruders and change the color of that extruder. When trying to manually put it in with the G code for M 600 it only defaults as well to the extruder that’s currently printing.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

ex-nerd, 2023-11-06:
Is this related to https://github.com/prusa3d/PrusaSlicer/issues/11516 ?

sith77, 2023-11-06:
Kind of but not really , right now I have a two tool unit. I have a piece for a customer trying to do has two colors at the start and as it proceeds further through the Print, I want to swap one of the colors and change it if I try to select a certain section of the model to input a change filament, it will only change the current primary printing extruder. There's no option to change the extruder that is not currently printing. If I try to input any of the G code processes in there it will only allow the G code to change the current printing Auder if I try to designate T0 or T1 it completely ignite the line of code altogether, and doesn't do anything. In the menus, while the machine is printing, there is an option to change filament, but it gives no designation to allow you to select which extruder you want to change the color in.

ex-nerd, 2023-11-06:
This is what I see (2.7 alpha, MacOS) when I try to add a color swap with my 2-tool:

[image]

sith77, 2023-11-07:
[image]

sith77, 2023-11-07:
This is the message I get when I tried to use that method and it does 1 of 2 things, the machine does a pause in the print on the current printing extruder or completely ignores the mark and just prints as normal

Jan-Soustruznik, 2024-01-04:
Hello, @sith77,

thank you for your report. We will check the problem. I closed this issue as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/11516

closed 2024-01-04 by Jan-Soustruznik