prusa3d/PrusaSlicer #11601: Prusa Crashes when slicing specific file

rhys-ender, 2023-11-06:
### Description of the bug

When slicing a file from VORON StealthBurner for Revo Micro using VORON recommended (and other) settings, Prusa Slicer crashes.
Seems to fail when generating infill, based on last information in GUI. I have tried 100% infill and get the same results

### Project file & How to reproduce

[stealthburner_printhead_revo_micro_rear_cw2.zip](https://github.com/prusa3d/PrusaSlicer/files/13263155/stealthburner_printhead_revo_micro_rear_cw2.zip)
Try to Slice project

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 10

### Printer model

Creality Ender 3 Original

rhys-ender, 2023-11-06:
Related to the Extrusion Width settings
Only crashes when introducing these changes.
[image]

lukasmatena, 2023-11-06:
Thanks. This is a duplicate of #11267, which is already fixed in 2.7.0-alpha1. Closing.

closed 2023-11-06 by lukasmatena