prusa3d/PrusaSlicer #11604: Crash when perimeter generator set to Classic (works with Arachne)

meermanr, 2023-11-06:
### Description of the bug

PrusaSlicer Version 2.7.0-alpha1+MacOS-arm64 crashes when I attempt to slice the attached project for my PrusaXL multi-tool (3x extruders), but only when I change the perimeter generator from the default Arachne to Classic.

### Project file & How to reproduce

1. Open the attached 3mf project: [pillbox-dual-color.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13267770/pillbox-dual-color.3mf.zip) (after unzipping)
2. Navigate to setting "Print Settings -> Layers and Perimeters -> Advanced -> Perimeter generator" and change it from Arachne to Classic
3. Attempt to slice (e.g. ⌘+R)
4. macOS crash dialog appears here, as pictured below:

[image]

Full content of text:

[PrusaSlicer crash Version 2.7.0-alpha1+MacOS-arm64.txt](https://github.com/prusa3d/PrusaSlicer/files/13267763/PrusaSlicer.crash.Version.2.7.0-alpha1%2BMacOS-arm64.txt)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0-alpha1+MacOS-arm64

### Operating system

macOS Sonoma 14.1

### Printer model

PrusaXL 5T (Alpha)

Jan-Soustruznik, 2025-04-02:
Hello @meermanr, Could you check with the latest PS release, version 2.9.1? The issue shoud be resolved already in 2.9.0-Alpha1, where multi-material segmentation for painted volume objects was improved. Please let us know. Thank you, and have a great day!

meermanr, 2025-04-02:
Yup, confirmed to work without crashing in Version: 2.9.1. Thanks for following up!

closed 2025-04-02 by meermanr