prusa3d/PrusaSlicer #11607: Bad/messy/inconsistent slicing in PS Alpha (entire layer, even perimeters, replaced with 100% infill?)

ex-nerd, 2023-11-06:
### Description of the bug

I don't really know how to describe this bug other than it creates an incredibly messy layer, like the whole thing is 100% infill instead of recognizing perimeters. The Z position of the messy layer changes (along with the angle of the "mess") if you move the part around on the build plate, and seems to go away completely if you rotate the part 180°

[image]

### Project file & How to reproduce

[messy_slice_layer_15.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13271215/messy_slice_layer_15.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1

### Operating system

MacOS 

### Printer model

Prusa XL 2-tool

rtyr, 2023-11-06:
There are some errors in the model. It is sliced correctly after repairing the model. I am attaching repaired model.

[fixed_messy_slice_layer_15.zip](https://github.com/prusa3d/PrusaSlicer/files/13271542/fixed_messy_slice_layer_15.zip)

ex-nerd, 2023-11-06:
Out of curiousity what was wrong with it and how did you fix it? I didn't get any warnings when importing the original model from Onshape, nor when opening this one. In the interest of checking whether this is a problem with Onshape's 3mf export or a bug with the "merge" functionality in PrusaSlicer, I've attached the raw export from OnShape:

[hsw_350x310_fat-left.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13271700/hsw_350x310_fat-left.3mf.zip)

I generated the "messy" one by importing this 2-part model (requires pressing `undo` because PS auto-arranges the 2 parts into different places … thankfully `undo` puts them back), and then selecting both parts from the list and then `merge` from the right-click menu.

FWIW, upstream info about this model (I'm the author) is at https://www.printables.com/model/620175-honeycomb-storage-wall-onshape-customizer (though if you want the specific part in question, it's the left+middle panel from [this break-out model with the measurements for my wall])https://cad.onshape.com/documents/cc454e61b35deaa13afb1e4a/w/4830c0683428b4b2a1e49227/e/4d789a6442cbe2f3de2b791a?configuration=default&renderMode=0&uiState=65494015d72e94791c6093bc))

ex-nerd, 2023-11-06:
@rtyr sorry, forgot to tag/quote you on that last reply so you'd see it.

rtyr, 2023-11-06:
I just repaired it using built-in windows repair service (it is based on Netfabb).

ex-nerd, 2023-11-06:
> I just repaired it using built-in windows repair service (it is based on Netfabb).

Interesting. Does it flag the same kind of errors in the `hsw_350x310_fat-left.3mf.zip` file I attached above? If so, then this is something with Onshape. If not, I'll edit the description here and re-classify it as a bug in PrusaSlicer's "merge" function.

ex-nerd, 2023-11-06:
Even stranger, I created a new PS project with the new 3mf export I did today, and it slices fine. I'll close this and just keep an eye on things in case it's some issue with PS, but I know Onshape has had issues in the past with 3mf export compatibility so it seems this is more likely something on their end.

ex-nerd, 2023-11-06:
Reopening after a bit more testing. This definitely seems to be a problem with how PS handles merged files. I've also confirmed that it still happens in `2.6.1` (Mac).

Replication steps:

If you take the original file I uploaded and split the `Merged` part back up into separate objects, everything slices fine. It also seems to slice properly if you set both parts to print with the same extruder. However, if you re-merge the parts, the bad layers return (though at different layer heights).

ex-nerd, 2024-04-28:
Still happening in the latest 2.7.4. Here's a new file that shows "no errors detected" and yet still exhibits the problems. You can see the bad layer at/around 4.0mm. However, the errors go away if you rotate the part 180° and re-slice. As with earlier issues the problem also goes away if you merge the two colors into a single part.

[bad-slicing-layer-bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/15144193/bad-slicing-layer-bug.3mf.zip)

ex-nerd, 2024-06-26:
Just checked. Still present in `2.8rc2`

ex-nerd, 2025-12-09:
Still present in 2.9.4. I just ran into this again with a new batch of HSW panels I was printing. No multicolor this time, so I've updated the description to reflect that. Fix is the same … rotate by some amount (90° time time) and the problem goes away.

ex-nerd, 2026-09-02:
Still present in PrusaSlicer 3:

[image]