prusa3d/PrusaSlicer #11609: ender 3 S1 pro no ABL command in start gcode

ghylander, 2023-11-07:
**Is your feature request related to a problem? Please describe.**
The current default profile for the Creality Ender 3 S1 Pro does not include the command to restore a saved bed mesh or perform an auto-leveling process

**Describe the solution you'd like**
Add the `M420 S1` (or alternatively `M420 S1 Z10`) command to enable ABL.

**Additional context**
Surprisingly, I've found issues and pull requests from 2022 supposedly fixing this issue, but as of today the start gcode includes neither the `M420` nor `G29` commands.

rtyr, 2023-11-07:
@pmjdebruijn fyi

ghylander, 2023-11-07:
Forgot to mention: I am aware that it is possible to build marlin with the "RESTORE_LEVELING_AFTER_G28" configuration.
However, either Creality does not use it or uses it only on some printers

I'm currently using the latest firmware (V2.0.8.28), but was experiencing this issue with the previous versions as well (V2.0.8.26 and V2.0.8.24)

Since using M420 just restores the existing ABL bed mesh, I don't think there is any issue if the command is executed on a printer with "RESTORE_LEVELING_AFTER_G28" enabled. The restored mesh would be overwritten with the same mesh.

pmjdebruijn, 2023-11-07:
I don't have any plans to change this really.

Currently there is consistent behavior between different Creality printers, and I'd like to keep it that way.

This also keeps bundle maintenance manageable.

pmjdebruijn, 2023-11-07:
I forgot the S1 series was again a special case, so if I understand correct you're not having any ABL now?

pmjdebruijn, 2023-11-07:
Did you try locally adding M420 S1 to your start_gcode ?

ghylander, 2023-11-07:
Hi, yes

I can provide some examples of prints using M420 S1 and prints not using it, but it'll have to wait a few days as i'm currently printing a very large and complex model

Besides pictures showing first layer quality, I can easily verify that when not using M421 S1, the Z axis does not move at all when moving around the plate

ghylander, 2023-12-05:
Hi,
As an update on this:

I'm currently trying to investigate the root cause because I'm not consistently getting the same results.
Sometimes, using M420 S1 will not use the mesh, and commenting it out will use the bed mesh
Sometimes M420 S1 will use the bed mesh, and commenting it out will not use it

My current hypothesis is:
Power cycling, updating firmware, etc the bed mesh will not work if using just G28
The first time the printer encounters a M420 S1 it will use the bed mesh
Subsequent prints will use the bed mesh if using G28, but will not use it if another M420 S1 is found

As I said I'm getting inconsistent results and I'm trying to perform tests between prints

ghylander, 2026-09-22:
Closing this myself
The issue stands but it's related to creality's buggy firmware, not the slicer

closed 2026-09-22 by ghylander