prusa3d/PrusaSlicer #11615: [FEATURE REQUEST] Include wipe tower in list of objects so it can be cancelled mid-print

DaveTSG, 2023-11-08:
**Is your feature request related to a problem? Please describe.**
I was using my XL today, printing several separate parts in two colours (some parts in white, some parts in black). The wipe tower detached from the plate mid-print, so I decided to cancel the printing of half of the objects, e.g. all of the black parts, so that it would only print all of the white parts. I thought this would remove the need for a wipe tower, and so the printer would automatically cancel printing on the (now removed) wipe tower. I was wrong.

**Describe the solution you'd like**
Make the wipe tower a separate part listed in the side-bar in PrusaSlicer. 

**Describe how it would work**
If the wipe tower is a part that is selectable in the side-bar in PrusaSlicer, it would also be a separate part in the Gcode that is generated, and would then be able to be cancelled using the new "Cancel object directly on the printer" feature in XL firmware v5.1.0-alpha2.

**Describe alternatives you've considered**
Cancelling the entire print was considered, but would've wasted filament, so wouldn't be ideal.
An alternative would be to add in code in the firmware that detects the wipe tower and allows it to be deleted from there, but I would imagine that would be far more difficult to program in than simply making the wipe tower a part in PrusaSlicer.

**Additional context**
Add any other context or paint drawing about the feature request here.

BrienKing, 2023-11-08:
Yes please.

KlausKraemer, 2024-02-17:
Another way would be a choice of a more solid tower in Prusa Slicer. The one and only tower is very flimsy, even though it often works. Compared to a big print gone wrong, having a little more waste should even be sustainable, before throwing the wrecked print away.

JoaquinBerrios, 2024-03-29:
Including the tower in the list of objects would not only be useful for cancelling the tower, but also when using adaptive leveling (i.e. KAMP in Klipper) so that the area under the tower is also probed. Currently, adaptive leveling does not probe under the tower since the tower is not included as an object and that has caused adhesion issues with the purge tower for me, so I had to disable adaptive leveling and revert to probing the entire bed which takes significantly longer and produces a lower resolution map.

KlausKraemer, 2024-05-18:
A highly needed feature! The purge tower is way too flimsy. At least cancelling the sh(beeep) would help losing prints. In the meantime I try to avoid using a purge tower at all.

Flyingvette, 2024-08-14:
The software architecture is already in place to let all this happen. It's nonsensical to not allow us to cancel it mid print as other objects are. As others have noted, has also led me to have to cancel multiple prints due to the wipe tower giving up on its relationship with the plate. In addition, a setting should be added for us to increase the depth of the wipe tower brim, or even better, allow for a variable ramping design from the edge of the brim to the wipe tower structure to decrease the likelihood of it letting go.

kurtgluck, 2025-07-28:
All that has to be done, is to 1) add the wipe tower to the list of objects at the beginning of the GCODE (perhaps as object 0, bumping all the other index’s).  Then bracket the code to actually extrude on the wipe tower with the start/end object gcode.

Tupson444, 2026-09-17:
Still relevant (unless it is implemented in 3.0.0, I didnt check)