prusa3d/PrusaSlicer #11617: Visual error from mistype in scaling tool

Lord-Emloy, 2023-11-08:
### Description of the bug

I was working on make a box to put under a trophy. I tried to hypothetically set the trophy on top and it didntwork so I was adjust the box size and then move the trophy to see if the size looked good. 

One of the times the key on my keyboard held down and instead of 78mm i put something like 788888888mm. When scaled back down it never corrected the program back to the previous state. hitting the undo keys also did not resolve the issue.

### Project file & How to reproduce

[Prusa Bug Report.zip](https://github.com/prusa3d/PrusaSlicer/files/13291888/Prusa.Bug.Report.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0-alpha1+win64

### Operating system

Windows 10

### Printer model

Prusa Mk4IS

rtyr, 2023-11-08:
I was not able to reproduce it.

FidelCapo, 2024-01-29:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2024-01-29 by FidelCapo