prusa3d/PrusaSlicer #11620: 2.7-alpha1 skips leading zero on G2 J/I parameters for legacy marlin machines (Snapmaker A350)

tyeth, 2023-11-08:
### Description of the bug

Machine doesn't like instruction that is missing leading zero.
e.g.
```G2 X186.82 Y226.366 I.037 J-.212 E33.22049```

### Project file & How to reproduce

[FrameAndVentPETG_A350.zip](https://github.com/prusa3d/PrusaSlicer/files/13300043/FrameAndVentPETG_A350.zip)
Printer pauses at inner walls here in mid section of model (90 degree bend) which appears to use the G2 move for inner walls.
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1

### Operating system

win11 64

### Printer model

Snapmaker A350

neophyl, 2023-11-08:
This is a duplicate issue.  Its been raised before. Please see https://github.com/prusa3d/PrusaSlicer/issues/7505  or https://github.com/prusa3d/PrusaSlicer/issues/6996

In particular this comment https://github.com/prusa3d/PrusaSlicer/issues/6996#issuecomment-1023408827

tyeth, 2023-11-08:
If a regex is needed to fix printer profiles that Prusa supply (yes for other manufacturers) to fix known faults, then include it in the profile. It's a bug, you just choose Won't fix.

neophyl, 2023-11-08:
Profiles for non Prusa machines are submitted by users. There’s another repo for those but I don’t know what the process is for submitting to it. I’ve seen other issues here where one of the devs has pinged someone else when profile issues come up for other machines though.

tyeth, 2023-11-08:
Thanks neophyl, I could and should have done this years ago. PR submitted after coping from my exported settings bundle.

rtyr, 2023-11-08:
>  Using official printer profile for Snapmaker A350 supplied by Prusa as part of PrusaSlicer 2.7.0-alpha1

G2/G3 commands are emitted only with arc fitting enabled. It is not enabled in bundled profiles and it may be not good idea to enable it for printers with old Marlin (various bugs in arc implementation).

Do you have any issues without arc fitting enabled?

tyeth, 2023-11-08:
I am trying to use arc fitting as it was recently added to the snapmaker
firmware.
I am reluctant to turn that on in the live profile, but the regex is an
"official" fix which resolves issues with certain firmware versions (it
happened before with G1 years ago so it's a general improvement for
snapmaker machines known to be potentially vulnerable to this issue).

rtyr, 2023-11-10:
PR was merged and distributed via configuration update 1.0.1.

closed 2023-11-10 by rtyr