prusa3d/PrusaSlicer #11621: A method for less visible seams

MichaelJLew, 2023-11-08:
The problem presented by the seams at layer change needs little introduction. It is sometimes trivial and sometimes important and PrusaSlicer offers a lot of variables that can alter the seam location, form, and visibility on the model. However, there is often no good way to prevent the seam from being  #intrusive.

I have a method for making seams in my 'hand-made' gcode that are far less intrusive than those that I can achieve using PrusaSlicer. I believe that the method would be readily incorporated into the slicer and should be offered as an option.

I call the improved seam a 'scarf seam' because it is similar to a scarf joint that woodworkers might use to join two pieces of timber end to end. The scarf joint is conceptually simple as it simply has matching tapers on the ends of the timber that are glued together. A long taper gives a strong joint.

The standard seam can be described like this: the nozzle is positioned at the start of a perimeter at the z-height that is a full layer height above the previous layer. Extrusion starts and the nozzle moves around the perimeter until it arrives back at the start position whereupon the nozzle movement and extrusion stops. The scarf seam differs in that the nozzle starts at the same z-height as the previous layer. The Z-height is gradually increased as the perimeter is printed until it reaches the full layer height. (That is one of the scarf tapers.) The perimeter is then printed normally until the nozzle reaches back to the starting XY location, whereupon the nozzle retraces the first taper with a complementary taper where the layer height is decreased gradually to zero but the z-height is fixed. (That makes the second taper.) After the second taper is complete the nozzle can be moved away to complete the rest of the layer in the normal manner. 

The low visibility of the scarf seam comes from the fact that the discontinuity of the joint if spread over the gradual tapers. When the nozzle is moved to the outer perimeter to start a scarf seam it is moved with effectively zero layer height and so there is almost no artefact from that movement. The scarf seam is compatible with any object shape, but I used a cylinder for the example for my own convenience. (I normally print fountain pen parts and most of my printing is pseudo-cylindrical objects with gcode from custom-written software and so I have a big library of cylinder-related functions.)

The scarf seam has a couple of possible disadvantages. First, it gives a region of variable layer height that may sometimes be visible. Second, it will be slower than the normal seam because it involves the nozzle moving twice over a part of the outer perimeter. That speed difference may be trivial in many cases as the scarf distance can be small compared to the perimeter total, and the outer perimeter is a small part of most printed objects. Finally, the scarf seam requires that the outer perimeter be printed first so that the first taper does not foul adjacent extrusions.

**Example file**

The attached file was generated by PrusaSlicer and then gcode generated by custom software appended. The PrusaSlicer settings were for the MK4 (no IS) and used the 0.20 QUALITY setting for Prusament PLA with the following changes from the defaults: no retract on layer change; no wipe; external perimeters first. The seam was painted on to force it to align with the seams from my software.

The first 6mm of the cylinder are the PrusaSlicer output. The region between 6mm and 12mm has a scarf seam with the tapers extending around the full 360 degrees of circumference. Between 12mm and 18mm the scarf seam has aligned tapers over 80 degrees of circumference. The alignment requires a retrograde travel move after each layer. The last section, 18mm to 24mm, has the 80 degree scarf but does not re-align the seams and so the seam moves by 80 degrees at each layer change. 

[image]

[cylinder d=12 h=6 v2d.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/13302344/cylinder.d.12.h.6.v2d.gcode.zip)

mix579, 2023-11-09:
This looks really promising and would make one of the banes of FDM less of an issue.

thanhtacles, 2023-11-18:
Very few new features in a slicer are revolutionary these days, but this seems like it could be one of them.

ulfertg, 2023-11-25:
I would love to see this idea receive more attention.
I've printed the test cylinder, and the result is simply great.

I can imagine that implementing it in a slicer might be quite complicated since an algorithm would have to handle various shapes and sizes. Nevertheless, the result is very compelling, at least visually.

MichaelJLew, 2023-11-25:
I think implementation would be easy in the slicer. It would just have to keep track of the first x millimetres of each layer, do the taper over those point and then complete the matching taper over the same points, in the same sequence, at the end of the layer.

I’m hoping that now that PS2.7 is out there will be some attention given to scarf seams…

MichaelJLew, 2023-11-26:
Yes, that would probably work. In fact, my own scarf code is adapted from a routine that I wrote to go between vase-mode and layer by layer portions of my objects.

I think that the best solution would involve both the slicer and the firmware because the firmware could have responsibility for tapering extrusions over the length of a move. Such a capability would be very useful for non-planar printing. I once asked for this feature for RepRap firmware in the Duet forum, but the response was that my use-case was too specialised.

MichaelJLew, 2023-12-03:
@popsickl3 Good idea. Yes, I'll post to Reddit, but not Facebook. 

I deleted my Facebook account after Zuck threatened the Australian federal government with retaliation for proposed legislation that would have Facebook paying a small amount to Australian commercial news outlets for their news. Facebook then closed down all of the government departmental facebook pages and those of many emergency and social support services! He/they are not, at the moment, as obviously odious as Musk, but he/they are equally deserving of no support from me. Sorry for the rant, but things like that should not be forgotten. You can read about it here: https://www.theguardian.com/media/2021/feb/23/facebook-reverses-australia-news-ban-after-government-makes-media-code-amendments

wafflecart, 2023-12-05:
> wow, that's astounding, I will peek at the code.

There is none, just an idea..

wafflecart, 2023-12-05:
> I think implementation would be easy in the slicer. It would just have to keep track of the first x millimetres of each layer, do the taper over those point and then complete the matching taper over [...]

If it's easy open a Pull Request with you code, it's just an idea at this point..

MichaelJLew, 2023-12-05:
> If it's easy open a Pull Request with you code, it's just an idea at this point..

I don't know what a Pull Request is. Yes, it's an idea for the slicer, but I have already implemented it in my own LiveCode functions, so it is not "just" an idea. I have posted an example of Gcode that works, so it is not "just" an idea. I have also given a description that is pretty much pseudo-code, so it is not "just" an idea.

Does a person have to provide C++ code in order to contribute? I don't think so, and I find your comment to be a bit offensive.

wafflecart, 2023-12-05:
> I don't know what a Pull Request is. Yes, it's an idea for the slicer, but I have already implemented it in my own LiveCode functions, so it is not "just" an idea. I have posted an example of Gcode [...]

I was referring to your previous comment saying "I think implementation would be easy in the slicer.". It's not that simple and could take a long time. Even more so without any requirements, code or pull request.

MichaelJLew, 2023-12-05:
I knew what you were referring to, but thank you for mansplaining it to me.

MichaelJLew, 2023-12-10:
I have been castigated on Reddit for saying that the implementation of this should be "easy". I guess I should have said that the implementation is conceptually straightforward. I have also been castigated for not supplying the code. I assumed that my Livecode code would not be helpful, but here it is:
`on mouseUp
   put 50 into xc
   put 50 into yc
   put 0.2 into LH
   put 0.45 into w
   put the cPrusaSlicerCode of me into tcode--6mm tall cylinder, two walls, LH=0.2
   
   put 6 into z
   put 6 into rout
   put rout-0.9 into rLumen
   put 0 into theta
   put 600 into f
   
   put rout-w/2 into r
   put unpolarGoToGC(xc, yc, r, theta, z, 0, f) & return after tcode
   repeat while z<=12
      put smallSeamLayerMW(xc, yc, rout, rLumen, LH, theta, z, f) & return after tcode
   end repeat
   
   put rout-w/2 into r
   put unpolarGoToGC(xc, yc, r, theta, z, 0, f) & return after tcode
   put 1.4 into seamDtheta
   put 0 into theta
   repeat while z<=18
      put smallSeamLayerMW2(xc, yc, rout, rLumen, LH, seamDtheta,theta, z, f) & return after tcode
      subtract seamDtheta from theta
   end repeat
   put unpolarGoToGC(xc, yc, r, theta, z, 0, f) & return after tcode
   repeat while z<=24
      put smallSeamLayerMW2(xc, yc, rout, rLumen, LH, seamDtheta,theta, z, f) & return after tcode
   end repeat
   put tcode into fld "gcodeFld"
end mouseUp

function smallSeamLayerMW xc,yc,rout,rLumen,LH, theta,@z,f
   put true into ccw
   put rout-rLumen into totW
   put max(1,totW div 0.45) into n
   put totW/n into w
   put rout-w/2 into r
   --outer perim made of two tapers
   put unpolarGoToGC(xc, yc, r, theta, z, 0, f) & return after tcode
   put circle1Taper(xc, yc, r, z, theta, w, LH, f, ccw) & return after tcode
   put circle1Taper(xc, yc, r, z, theta, w, LH & return & 0, f, ccw) & return after tcode
   --inner perims
   --put r into ri
   repeat n-1
      put r-w into r
      put unpolarGoToGC(xc, yc, r, theta, z, 0, f) & return after tcode
      put arc1(xc, yc, r, 0, 2*pi, z, theta, w, LH, f, ccw) & return after tcode
   end repeat
   --   --return to start
   --   put unpolarGoToGC(xc, yc, r, theta, z, 0, 1.5*f) & return after tcode
   return tcode
end smallSeamLayerMW

function smallSeamLayerMW2 xc,yc,rout,rLumen,LH,seamDtheta,@theta,@z,f
   put true into ccw
   put rout-rLumen into totW
   put max(1,totW div 0.45) into n
   put totW/n into w
   put rout-w/2 into r
   put unpolarGoToGC(xc, yc, r, theta, z, 0, 2000) & return after tcode
   put ";taper 1" & return after tcode
   put 0 into LH1
   put LH into LH2
   put arc1taperLH(xc,yc,r, LH,seamDtheta,z,theta,w,LH1,LH2,f,ccw) & return after tcode
   put return after tcode
   put arc1(xc,yc,r, 0,2*pi-seamDtheta,z,theta,w,LH,f,ccw) & return after tcode
   --need a taper here that decreases LH by keeps z constant...
   put LH into LH1
   put 0 into LH2
   put ";taper 2" & return after tcode
   put arc1taperLH(xc,yc,r, 0,seamDtheta,z,theta,w,LH1,LH2,f,ccw) & return after tcode
   put return after tcode
   --inner perims
   --put r into ri
   repeat n-1
      put r-w into r
      put unpolarGoToGC(xc, yc, r, theta, z, 0, 2000) & return after tcode
      put arc1(xc, yc, r, 0, 2*pi, z, theta, w, LH, f, ccw) & return after tcode
   end repeat
   --   --return to start
   --   put unpolarGoToGC(xc, yc, r, theta, z, 0, 1.5*f) & return after tcode
   return tcode
end smallSeamLayerMW2

function arc1taperLH xc,yc,r, dz,dtheta,@z,@theta,w,LH1,LH2,f,ccw
   --Uses only G1 commands so that it can work with Prusa MK4 firmware.
   --Can taper w, but LH is constant. Use circle1Taper to vary LH.
   if ccw is empty or ccw="ccw" then put true into ccw
   if dtheta="dtheta" then put 2*pi into dtheta
   put LH1 into LH
   put line 1 of w into w1
   put line 2 of w into w2
   if w2 is empty then put w1 into w2
   put round(6*(r+1)^1.5) into numSegsPerCircle
   put round(numSegsPerCircle*dtheta/(2*pi)) into numSegs
   if numSegs<1 then
      return "; arc1 had less than  one segment; r=" & r
   end if
   put abs(dtheta) into dtheta --can't have negative here. Use ccw=false to go clockwise.
   put dtheta/numSegs into dtheta
   put dz/numSegs into dz
   put (w2-w1)/numSegs into dw
   put line 1 of w into w
   put (LH2-LH1)/numSegs into dLH
   if not ccw then put -dtheta into dtheta
   put unpolarX(xc, r, theta) into x1
   put unpolarY(yc, r, theta) into y1
   add dtheta to theta
   put unpolarX(xc, r, theta) into x
   put unpolarY(yc, r, theta) into y
   put sqrt((x1-x)^2+(y1-y)^2) into dist
   repeat numSegs
      add dz to z
      add dw to w
      add dLH to LH
      put extrudeAmount(dist, LH, w, 1.75) into e
      if e<0 then
         return "; extrusion less than zero; r=" & r
      end if
      put G1line(x, y, z, e, f) & return after tcode
      add dtheta to theta
      put unpolarX(xc, r, theta) into x
      put unpolarY(yc, r, theta) into y
   end repeat
   subtract dtheta from theta
   return tcode
end arc1taperLH
`

MichaelJLew, 2023-12-10:
Seems that the Code function plays badly with the LiveCode code, so here it is again (with a couple of commented out lines removed):

("--" denotes a comment; code is case-insensitive; indentation is for clarity only)

on mouseUp
   put 50 into xc
   put 50 into yc
   put 0.2 into LH
   put 0.45 into w
   put the cPrusaSlicerCode of me into tcode--6mm tall cylinder, two walls, LH=0.2
   
   put 6 into z
   put 6 into rout
   put rout-0.9 into rLumen
   put 0 into theta
   put 600 into f
   
   put rout-w/2 into r
   put unpolarGoToGC(xc, yc, r, theta, z, 0, f) & return after tcode
   repeat while z<=12
      put smallSeamLayerMW(xc, yc, rout, rLumen, LH, theta, z, f) & return after tcode
   end repeat
   
   put rout-w/2 into r
   put unpolarGoToGC(xc, yc, r, theta, z, 0, f) & return after tcode
   put 1.4 into seamDtheta
   put 0 into theta
   repeat while z<=18
      put smallSeamLayerMW2(xc, yc, rout, rLumen, LH, seamDtheta,theta, z, f) & return after tcode
      subtract seamDtheta from theta
   end repeat
   put unpolarGoToGC(xc, yc, r, theta, z, 0, f) & return after tcode
   repeat while z<=24
      put smallSeamLayerMW2(xc, yc, rout, rLumen, LH, seamDtheta,theta, z, f) & return after tcode
   end repeat
   put tcode into fld "gcodeFld"
end mouseUp

function smallSeamLayerMW xc,yc,rout,rLumen,LH, theta,@z,f
   put true into ccw
   put rout-rLumen into totW
   put max(1,totW div 0.45) into n
   put totW/n into w
   put rout-w/2 into r
   --outer perim made of two tapers
   put unpolarGoToGC(xc, yc, r, theta, z, 0, f) & return after tcode
   put circle1Taper(xc, yc, r, z, theta, w, LH, f, ccw) & return after tcode
   put circle1Taper(xc, yc, r, z, theta, w, LH & return & 0, f, ccw) & return after tcode
   --inner perims
   repeat n-1
      put r-w into r
      put unpolarGoToGC(xc, yc, r, theta, z, 0, f) & return after tcode
      put arc1(xc, yc, r, 0, 2*pi, z, theta, w, LH, f, ccw) & return after tcode
   end repeat
   return tcode
end smallSeamLayerMW

function smallSeamLayerMW2 xc,yc,rout,rLumen,LH,seamDtheta,@theta,@z,f
   put true into ccw
   put rout-rLumen into totW
   put max(1,totW div 0.45) into n
   put totW/n into w
   put rout-w/2 into r
   put unpolarGoToGC(xc, yc, r, theta, z, 0, 2000) & return after tcode
   put ";taper 1" & return after tcode
   put 0 into LH1
   put LH into LH2
   put arc1taperLH(xc,yc,r, LH,seamDtheta,z,theta,w,LH1,LH2,f,ccw) & return after tcode
   put return after tcode
   put arc1(xc,yc,r, 0,2*pi-seamDtheta,z,theta,w,LH,f,ccw) & return after tcode
   --need a taper here that decreases LH by keeps z constant...
   put LH into LH1
   put 0 into LH2
   put ";taper 2" & return after tcode
   put arc1taperLH(xc,yc,r, 0,seamDtheta,z,theta,w,LH1,LH2,f,ccw) & return after tcode
   put return after tcode
   --inner perims
   repeat n-1
      put r-w into r
      put unpolarGoToGC(xc, yc, r, theta, z, 0, 2000) & return after tcode
      put arc1(xc, yc, r, 0, 2*pi, z, theta, w, LH, f, ccw) & return after tcode
   end repeat
   return tcode
end smallSeamLayerMW2

function arc1taperLH xc,yc,r, dz,dtheta,@z,@theta,w,LH1,LH2,f,ccw
   --Uses only G1 commands so that it can work with original Prusa MK4 firmware.
   --Can taper w, but LH is constant. Use circle1Taper to vary LH.
   if ccw is empty or ccw="ccw" then put true into ccw
   if dtheta="dtheta" then put 2*pi into dtheta
   put LH1 into LH
   put line 1 of w into w1
   put line 2 of w into w2
   if w2 is empty then put w1 into w2
   put round(6*(r+1)^1.5) into numSegsPerCircle
   put round(numSegsPerCircle*dtheta/(2*pi)) into numSegs
   if numSegs<1 then
      return "; arc1 had less than  one segment; r=" & r
   end if
   put abs(dtheta) into dtheta --can't have negative here. Use ccw=false to go clockwise.
   put dtheta/numSegs into dtheta
   put dz/numSegs into dz
   put (w2-w1)/numSegs into dw
   put line 1 of w into w
   put (LH2-LH1)/numSegs into dLH
   if not ccw then put -dtheta into dtheta
   put unpolarX(xc, r, theta) into x1
   put unpolarY(yc, r, theta) into y1
   add dtheta to theta
   put unpolarX(xc, r, theta) into x
   put unpolarY(yc, r, theta) into y
   put sqrt((x1-x)^2+(y1-y)^2) into dist
   repeat numSegs
      add dz to z
      add dw to w
      add dLH to LH
      put extrudeAmount(dist, LH, w, 1.75) into e
      if e<0 then
         return "; extrusion less than zero; r=" & r
      end if
      put G1line(x, y, z, e, f) & return after tcode
      add dtheta to theta
      put unpolarX(xc, r, theta) into x
      put unpolarY(yc, r, theta) into y
   end repeat
   subtract dtheta from theta
   return tcode
end arc1taperLH

MichaelJLew, 2024-01-30:
This is now being implemented for OrcaSlicer: https://github.com/SoftFever/OrcaSlicer/pull/3839

mrflib, 2024-02-10:
I never really noticed this was an issue before. I started recently designing (with a view to selling) astronomy parts. Telescopes are round and cylindrical for the most part. These seems are bad enough that I'm not actually sure I can sell them in the current state! 

This solution does seem fantastic. I wonder, does it work with lower layer heights than 0.3? I have a details on cylinder walls which benefit from lower layer height, around 0.15mm.

I hope Prusa actually acknowledge the effort put in by the community in this thread.

MichaelJLew, 2024-02-10:
Yes, it works well for me with layers of 0.12mm. I'm not sure, but the micro-step distance on the z-axis might be a limiting factor...

Are you aware that Noisyfox has implemented the method in OrcaSlicer? See this thread:
https://github.com/SoftFever/OrcaSlicer/pull/3839
You can try it out here: https://github.com/SoftFever/OrcaSlicer/actions/runs/7792877482

mrflib, 2024-02-17:
I saw that. I've not looked in to alternative slicers, so will take a look thanks. I was on with Prusa Support the other day and I linked this thread, he said he would forward it on. Believe it or not the seams were actually interfering with my interference fit. I really hope they implement it. It's great.

mrflib, 2024-02-18:
So I had a go with the Orca Slicer implementation. I do not have Orca dialled in at all for my printer, and I simply used default settings. As expected in such a scenario, PrusaSlicer gave a much nicer finish generally out of the box. However, the seam is absolutely outrageous. 

Orca Slicer's implementation of the Scarf Seam seems good. Printing allows you to see what most of the settings represent. I would say that it certainly minimises the stark impact of the seam, but does spread the seam over a much wider space, which is still visible. 

I think usage will really depend on what you're printing, as always.

None of these settings are dialled in, I'm sure I could get much better with experimenting as these are just default. But I figured sharing defaults might be useful 

### Prusa Slicer Prusament PLA Profile 0.4 IS Mk4 0.2mm Layer Height
Print Time 2hr 21min

[image]

### Orca Slicer Default Prusa PLA Profile 0.4 IS Mk4 0.2mm Layer Height
Print Time 3hr 29min (Default Scarf Seam settings)

[image]

MichaelJLew, 2024-02-18:
The scarf seam seems (sorry!) to have given you an improvement, but there is still room for more. Did you print the outer perimeter first? I would do that if the outer wall seam is the one you want best hidden. And I would not have retractions or z-hop between layers.

MichaelJLew, 2024-02-18:
> Print Time 3hr 29min

I just noticed the increased time for the scarf seam print. It seems to be too much. At most the scarf seam should extend print time by adding overlap on the perimeter and by (sometimes) requiring the outer perimeter to be printed slowly. That should not increase the time by anything like 50%.

Can you zip your gcode files and post them?

mrflib, 2024-02-18:
[E.zip](https://github.com/prusa3d/PrusaSlicer/files/14324495/E.zip)

Sure. As I said I just ran pretty much defaults on Orca. I'm sure it can be tweaked.

MichaelJLew, 2024-02-18:
> Sure. As I said I just ran pretty much defaults on Orca. I'm sure it can be tweaked.

I printed the bottom few mm of your files and got the results that you did (Prusa MK4, like yours). I then sliced a similar tube shape with retractions turned off and got a much better scarf seam, at the cost of having quite a few blobs on the inner surface and a little stringing. I think that turning off retract on layer change might help. I'll test that next.

mrflib, 2024-02-19:
Michael if you'd like the stl I used it's here:

https://drive.google.com/file/d/1WnE1i8HTZmbk9CAEqrja8WseL_imOmHK/view?usp=drivesdk

astraiophos, 2024-03-01:
+1 for adding this as a seam option to the slicer

Bappy1988, 2024-03-01:
+1 Just seen a video for this on TeachingTech YouTube channel. 100% want to see this in prusa slicer :) seams are the bane of my prints (well other than the printer itself that is!)

LouMinartey, 2024-03-02:
This looks incredible. +1 from me

KlausKraemer, 2024-03-02:
Another +1! This should be prioritized as it all but eliminates one of the ugliest issues of FDM printing.

Ruakij, 2024-03-02:
Please just add a Like to the first post instead of posting "+1", it gets harder to communicate with all that.

Zogar89, 2024-03-02:
Video about this new feature

https://www.youtube.com/watch?v=vl0FT339jfc

Androxilogin, 2024-03-03:
I was just watching a video on the **[Teaching Tech](https://www.youtube.com/watch?v=vl0FT339jfc)** YouTube channel about this the other day hoping for it to come to Prusa. This would be awesome. I've had nothing but problems with seams (and supports) using my Kobra 2 Pro.

faith-healer, 2024-03-04:
@petrubecheru wikipedia's administration rolled that back...

oursomiel, 2024-04-05:
This looks brilliant! Given the hype Orcaslicer put on it (well-deserved), I would hope to see it on PS very shortly!

Harvie, 2024-04-12:
Orca slicer is based on prusa slicer right? I guess it should be possible to backport the code...

oursomiel, 2024-04-12:
It is a mix of Bambu and prusa. I believe the base is Bambu

idaadi, 2024-04-27:
Is there a way on voting on such features ? anyone know ? @SachCZ

KlausKraemer, 2024-04-28:
> Is there a way on voting on such features ? anyone know ? @SachCZ

Give it a 👍 in the first item. At least that shows the interest and support.

RubenatorX, 2024-04-28:
@idaadi
> Please just add a Like to the first post instead of posting "+1", it gets harder to communicate with all that.

ShakieVan, 2024-06-16:
I can't wait to have this feature in PrusaSlicer (or alternatively a XL-Profile in CURA)!
It would resolve the biggest issue on functional "print in place"-prints.
When will it be implemented?

Steve7235, 2024-07-03:
I can't believe how long this idea has gone unnoticed for. This seems pretty easy to implement and would be a massive upgrade for PrusaSlicer. Maybe in 2.9?

MichaelJLew, 2024-07-03:
> has gone unnoticed

Well, it was noticed by OrcaSlicer, as it has been implemented in 2.0 and 2.1. I was surprised by the lack of response from the PrusaSlicer developers, but it no longer matters to me.

Androxilogin, 2024-07-04:
> > has gone unnoticed
> 
> Well, it was noticed by OrcaSlicer, as it has been implemented in 2.0 and 2.1. I was surprised by the lack of response from the PrusaSlicer developers, but it no longer matters to me.

I just installed Orca to try this. What is it listed as? I'm also wondering if the Kobra 2 profile would have the same settings as the Kobra 2 Pro.. There was no option for it.

MichaelJLew, 2024-07-04:
Scarf joint seam in the seam portion of the strength tab. Try it with the inner/outer/inner option for wall printing order if you have three or more perimeters.

Androxilogin, 2024-07-04:
Thanks, I'll give it a shot.

mrflib, 2024-07-04:
It's odd, I found some old rollers I printed years ago. They are 25cm long
and 4cm in diameter and printed vertically. There is almost no seam joint
at all.. the print is nearly flawless.

This would have been printed on a MK3 using prusaslicer.

I wonder why things have got worse

Harvie, 2024-07-04:
> I wonder why things have got worse

what was the layer height?

mrflib, 2024-07-04:
0.2 is all I did back then would have been completely stock prusa settings.
Fillamentum petg

sirguigui, 2024-07-17:
This seems very interesting, is there any way to currently use this in PrusaSlicer with an external post processing code? 

I am hoping that since this is already implemented in Orca, that the developers will also seriously consider it for PS.

klucky, 2024-07-30:
Such a great idea. I can't understand why it hasn't been implemented by the prusa team.

MrDixioner, 2024-08-01:
> Well, it was noticed by OrcaSlicer, as it has been implemented in 2.0 and 2.1. I was surprised by the lack of response from the PrusaSlicer developers, but it no longer matters to me.

You did a lot of work on this proposal: you described how it should work, you provided images and a file for the slicer, you provided the code that implements this function, but you only received complete ignorance from the developers :(
I fully support your suggestion and I would give it a thousand likes if it were possible!
I would like to know the reason for the complete lack of a solution to this problem on the part of the developers and hear their answer.

amatulic, 2024-08-01:
> I would like to know the reason for the complete lack of a solution to this problem on the part of the developers and hear their answer.

I think it's because the developers are (a) short-staffed, and (b) busy working on bigger changes and shiny new features, rather than algorithmic improvements to existing features, like the method described in this ticket. There's more to a fix than simply making sure it compiles without error and works; a huge matrix of use cases must also be tested, and this takes a lot of time.

I actually submitted a trivially simple PR (#9737) well over a year ago that fixed a bug in how temperature is controlled when switching between materials with the MMU, and while it got a lot of likes and positive comments like this issue, and Prusa staff were pinged multiple times, it never got a response, so it's still there. I and many others are using this fix, but we all have to compile PrusaSlicer ourselves to do so.

RubenatorX, 2024-08-01:
Ya'll need to calm down with the rage comments. They are aware of the feature:
<https://www.reddit.com/r/prusa3d/comments/1bjqeb1/comment/kvvhgyn/?context=1>

mix579, 2024-08-01:
They know about it but it's probably not on the top of their priority list. Importance is in the eye of the beholder. I for one would love a supported method for drawing st

bkerler, 2024-08-24:
I really would like to see this as well and I'm confident they are already working on this :)

ZoomDomain, 2024-09-16:
66 comments. my 1 kilo robot planetary gear was kinda ruined by seams. I have to change to orca slicer to attempt it again. thanks!

lukasmatena, 2024-11-21:
Implemented in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena

Harvie, 2025-02-08:
Thank you, it really makes a difference. When i enable it, the seam is still kinda visible, but when i run my fingers over it, it's way less noticeable.

BUT i need to point something out. On bowden printer it really helped me to also increase "seam gap distance". I was trying to push things to limits with 0.6mm nozzle and 0.44mm layer height, because i've noticed higher the layer height worse the seam gets. So i really wanted to see how the scarf joint performs under worst-case scenario. I've changed seam gap distance from 15% to 200% (which is rather lot, but i wanted to see what would happen - on bowden printers it does not really cause underextrusion, because there is still pressure in bowden). Now the seam is no longer protruding to the outside of the part and it's more embedded inside (therefore not messing up tolerances where multiple parts need to fit together). So i am pretty sure i can fine tune this gap in a way that it will be pretty much flush with surface. Also at 0.2mm layer height the seam is significantly less noticeable than on higher layer heights (the overall print quality difference between 0.2mm and 0.24mm is almost mindblowing to me).

**So to conclude: Scarf joint seam is pretty cool and exciting feature. But don't forget there are many other less exciting variables that do and will affect your seams greatly. Make sure to pay attention to them as well if seams are important for you. Especialy layer height, seam gap, and for bowden guys also all the pressure advance stuff and retractions**

Harvie, 2025-02-08:
Also please note that when you don't enable the "scarf seam entire loop" feature checkbox, it will actualy cause the slicer to create two seams instead of one. And yes, they are actualy noticeable in the final print. (unless your perimeter loop is shorter than "scarf seam length" setting)

[image]

TomLebeda, 2025-02-23:
Would it be possible to add an option (checkbox) for disabling the scarf seam only on the first layer, please? 

I have noticed that the very thin beginning of the transition sometimes doesn't stick well to the bed. Especially when the transition is starting smoothly from 0% and the transition length is longer.

amatulic, 2025-02-23:
I agree with @TomLebeda. No need for a checkbox though. Scarf seam should be disabled on the first layer always.

This is the most critical layer of any part, and you don't want to be doing weird things with the perimeter. You *especially* don't want to be reducing z height on the first layer in case the printer is set to squish that first layer deeply, else you risk scraping the nozzle against the build plate. I see absolutely no reason to have a scarf seam on the first layer.

lukasmatena, 2025-02-27:
@TomLebeda @amatulic 
Thanks for pointing this out. The scarf seams are now implicitly disabled on the first layer (since in [2.9.1-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-beta1)).