prusa3d/PrusaSlicer #11623: [BUG] 3.7.1-alpha1 - Incorrect GCODE compression / decompression cause crash on the MK4

antimix, 2023-11-09:
### Description of the bug

**PRINTER:**  MK4
**FIRMWARE:** 5.1.0-alpha2
**SLICER:** PrusaSlicer 2.7.0-alpha1

The print suddenly stop/crashed at layer 1 after few shapes. The printer looked like it was  printing on the LCD, but in reality it was doing nothing, head stopped, no movements, no messages special messages on the LCD, the scrolling printing as usual...

After further investigation the result was:
- the gcode that was sent to the printer via wifi with PrusaSlicer was correct. A binary compare with the one on the PC confirmed it was sent correctly.
- However going in step by step nozzle emulation on PrusaSlicer gcode, showed that at the precise point where it stopped (he had drawn very few shapes, so it was easy to find)  the uncompressed GCODE was weird, and this cause the MK4 to mess up.
- So PrusaSlicer produced bad gcode (either bad, and then bad compressed, or good, but bad compressed and bad decompressed). 
- Offending line: 476 
- There are probably others. 
- I did not have any issue in weeks with other print on the MK3S, so the issue is affecting the compression.

Attached the GCODE, and the PROJECT (is the MMU3 spool buffer)

[BUFFER_PART1_0.4n_0.2mm_PETG_MK4IS_3h23m.zip](https://github.com/prusa3d/PrusaSlicer/files/13307245/BUFFER_PART1_0.4n_0.2mm_PETG_MK4IS_3h23m.zip)
[image]

### Project file & How to reproduce

1) Open the project
2) Slice

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1

### Operating system

Windows 10

### Printer model

MK4

rtyr, 2023-11-09:
I printed several layers of your bgcode on MK4 without issues.

I also don't see any issue with the line 476 you are reffering to. G2 commands with I J parameters are normal if arc fitting is enabled (it is enabled by default in 2.7.0-alpha).

The issue you are describing is likely caused by known issue of FW 5.1.0-alpha2 (occassional USB disconnection during the print) or by some issue related to g-code streaming from PrusaConnect. Similar thing can also happen in case the filesystem on the usb drive is corrupted.

closed 2023-11-09 by rtyr

antimix, 2023-11-09:
@rtyr Thanks . In the meantime I have uncompressed the GCODE and reprinted it. It was printed without issue. So is was not an issue of the Slicer.