prusa3d/PrusaSlicer #11626: Uploading a print to PrusaLink with negative volume shows the whole print.

BitSmith99, 2023-11-10:
### Description of the bug

Need to test before committing to a large print?  I use [Negative Volume] to block out most everything except a test area.
PrusaLink shows the whole project, and does not indicate the Negative Volume
[image]
[image]

### Project file & How to reproduce

I just want to print a small test area to ensure the right thickness for each color.
[Sign Frame TEST.zip](https://github.com/prusa3d/PrusaSlicer/files/13315424/Sign.Frame.TEST.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.61+Win64

### Operating system

WinDoze 10 Home

### Printer model

MK4: 5.01+12089

kubispe1, 2023-11-10:
Right, the thumbnail generated by G-Code does not include negative volumes. You will see the same for the preview:
- On printer
- PrusaConnect 
- and so on ....
Closing this issue.

closed 2023-11-10 by kubispe1

BitSmith99, 2023-11-10:
Not good to summarily close the issue!  We want to see what the item looks like.  This is a good case to provide a toggle option to see either the Whole project or Printed object.