prusa3d/PrusaSlicer #11627: 2.7.0 alpha 1 crashes on my macbook when modifying purge volumes

nsheridan, 2023-11-10:
### Description of the bug

Prusa Slicer 2.7 alpha1 on a Macbook M2 Pro running MacOS Sonoma 14.1.1

In 2.7 alpha 1, if I change any of the purging volume values for a MMU print I get a crash.

[image]

[image]

### Project file & How to reproduce

1. Open Prusa Slicer 2.7 alpha1
2. Click on "purging volumes"
3. Click on an arrow to change any of the values.
4. Program crashes with a segmentation fault

[crash.log](https://github.com/prusa3d/PrusaSlicer/files/13320926/crash.log)
[repro.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13320919/repro.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7 alpha 1

### Operating system

macos Sonoma 14.1.1

### Printer model

Prusa mk3s+ with MMU2S

nsheridan, 2023-11-11:
This issue is still present in beta 1

YuSanka, 2023-11-13:
Thanks for your report.
SPE-2027

lukasmatena, 2023-11-16:
Fixed in [2.7.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc1). Closing.

closed 2023-11-16 by lukasmatena