prusa3d/PrusaSlicer #11628: I KNOW this is the wrong place...  I'm a newbee and don't know where to post GITHUB bugs.

BitSmith99, 2023-11-10:
### Description of the bug

Yes, I'm new here.  AND I've been searching, but haven't found the way to report a Github bug.  Simple issue; Dates shown are missing the year, so it's impossible to tell when something happened.

Here are a few;
[image]

### Project file & How to reproduce

I posted a Prusa Slicer bug, and someone pointed out that it's a repeat.  I guess we're the squeaky wheels to get things fixed.  S/he referenced previous posts, and I saw no way to tell when they were from.
**Solution:** Include a full date; month, day **and year**.
[image]

Also, please note GREY text on White background is VERY hard to read (I don't need "high contrast."  High Contrast should _only_ be used on gym memberships, car leases or other places where they **don't** want you to be able to read it.

**HOW do I post a bug report to Github so it can be fixed?**

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

n/a

### Operating system

n/a

### Printer model

n/a

lukasmatena, 2023-11-10:
The full time/date information shows on hover.
[image]

Closing, not related to PrusaSlicer.

closed 2023-11-10 by lukasmatena

BitSmith99, 2023-11-10:
> The full time/date information shows on hover. [image]
> 
> Closing, not related to PrusaSlicer.

_Yes, I said that.  But closing without info is not helpful._  **How do I report a bug?**