prusa3d/PrusaSlicer #11632: Sorely needed bridge option.

Strilock, 2023-11-10:
We lack proper settings for getting perfect bridges. The reason is because of complex and incorrect calculations that the slicer is performing for different layer hights. The slicer treats a bridge as regular infill calculating different width and height as if it is able to squish each layer. This is incorrect and this is discussed in the article flow math by Gary Hodgson in the original Sli3er manual ( https://manual.slic3r.org/advanced/flow-math ). Hence “thick bridges”. Thick bridges basically assumes round extrusion of nozzle’s diameter. However there is a big mistake in the original article, **overlap does not drag the filament**. In fact overlap causes the bridge to fuse into a single layer which prevents bulging and sagging (with PETG and PLA as tested). **But the flow(amount of material extruded)  will vary by layer hight!** This should not happen as a bridge does not really have a layer hight, it cannot be squished. That is the reason why floating help window says thick bridges may look worse but are more reliable. Thick bridges do not do that, they always extrude the same amount of filament, but it will never overlap as per original creators’ ideas. 

Now here is the difference. Here is the image of a single layer bridge with overlap from both sides.
[image]
[image]

Here is the image of thick bridge and how it looks at its best.
[image]

Overlapped bridge is printed in a single layer, it cannot bulge or sag or drip in between layer lines. Not sure how well it is visible on the pictures, but it is a single layer print, where as thick bridge with no overlap will sag always, because the extruded filament will be squeezed in between infill and the corners will never be crisp.

**The problem:** so the issue is that this type of printing is only possible because of a certain flow ratio + overlap. Thick bridges do not have overlap so they’re out automatically. At 0.16 layer hight its almost x2 for PETG. The slicer cannot go past x2 which is a problem for every layer below 0.16. The rest cannot fuse together and very low layer hights just tend to tear.

**Possible solutions:**
There are a few. First and I think is the easiest to implement is just put a tick box to have overlap for thick bridges (possibly controllable percentage of overlap). The rest can be controlled with flow and the bridges would be perfect. Another solution would be to allowing flow values past x2 but it is a bad solution since that would require calculating proper flow rates for perfect bridges for every layer hight. Potentially multi layer bridge settings can also help resolve bulging issue by limiting flow and increasing cooling for 1-2 layers above the bridge feature, but that is best implemented as an additional feature.

MichaelJLew, 2023-11-27:
This looks important and deserves some attention.

I'm not sure that I understand exactly what you have done for the examples in the photos. How did you get 'overlap' in the first object? (Why does it say "2x flow 1.6" on it?) My guess is that you did it by setting extra flow for the bridging extrusions. Perhaps you could say which settings you used in the slicer, and upload example gcode.

What i think that you are showing is that the bridging works well when the paths are placed close enough to overlap, no matter what the layer height might be. They fail without overlap and fail badly at layer heights below 0.16mm.

Strilock, 2023-11-28:
Well I thought so too but I do not think this will be implemented, there are a lot of bridge bugs in current release and they haven’t been fixed for 2 releases now.

You understood everything exactly right. 

Overlap is applied automatically by the slicer for 100% infill features. When you deselect “thick bridges” option, bridges are treated as any other infill. So there will be overlap by default. I’ve achieved this exactly as you thought I did, I increased bridge flow by 2.

Disregard the writing on the model, I did it for internal purposes, I did about 170 of them. This was meant to say 0.16 layer hight 2x flow.

However like the original author of the Sli3er article “flow math” noted, bridges are not like any other infill. That is correct, they are not, they do not have a width like any other infill type(they have hight but only kind of). However he said that overlap in it will cause the filament to be pulled. That is 100% incorrect. For ABS maybe since that was what they maybe printed when Slic3er was maintained. For PETG certainly this is not the case and not for PLA either.

So “thick bridges” option is the carry over from Slic3er. It is correct in everything but overlap. Prusa’s “no thick bridges” is incorrect in everything but overlap. We need to merge the two. The help menu even says for prusa's "no thick bridges" “look better but are less reliable” the only reason they are less reliable is because given settings only work for a given layer hight.

Now here are the pictures of thick bridges as best as you can make them and normal bridges with overlap. Those are pictures of the model with a few layers above the bridge. 

Here is a bridge with overlap:
[image]

And here is without:
[image]

Now I think it is clear that the majority would prefer option A.

What you can’t see though is that on an uncovered side, you would still be able to see the bridge sag a bit. Even the single layer one. This is perfect for bridging with covered sides but for exposed ones you will see a mm or 2 of sag. To clear that we need bridge layer 2 layer 3 settings. To increase cooling and decrease flow.

Timothy-Ecc19, 2026-09-02:
If better bridging options are not already in the plans for PrusaSlicer 3.0 these thoughts may still be relevant.