prusa3d/PrusaSlicer #11633: Bridge bug.

Strilock, 2023-11-10:
### Description of the bug

There is a rather serious bug currently present in Prusa Slicer that causes failed bridges on thin models. I’ve seen a lot of issues posted about bridges failing without listing the cause of failure.

Here is the preview of how Prusa Slicer would slice a thin model out of the box:
[image]

There is no anchorage for the bridge to grab on to so what will happen is something like this:
[image]

This is default behavior. The cause of this is the extra added perimeters (3 by default). For some reason extra perimeters give themselves the priority over bridge anchorage. I don’t think this is desired behavior since it causes failed prints. Same happens with top infill if the bridge is on the very top layer. Top infill gives itself priority over bridge anchorage. Although more rare it is still not a desired effect. It also does not have a workaround like reducing extra perimeters for thin models with bridges. The workaround works sometimes if it is possible to reduce the perimeters like so:
[image]
[image]
[image]

### Project file & How to reproduce

[BRIDGE_-_One-layer_bridging_test.zip](https://github.com/prusa3d/PrusaSlicer/files/13323238/BRIDGE_-_One-layer_bridging_test.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1. GTK 3

### Operating system

Linux Kubuntu 22.04

### Printer model

modified ender 3

loskexos, 2023-12-15:
Having trouble with ugly bridges in prints I went back to basices and 
tried to calibrate bridging for the filament in use.

I troubleshooted and ended up with what looks to be same issue as this thread.

One Layer Bridge. 
Ends up printing both ends with little to no touching points to anything solid. 
So layers all fall to printbed during print. 

simple model to reproduce attached (need to rotate upside down, for 0.2mm layer height - one object has 0.2mm bridge, the other 0.3mm bridge)
[bridgetester v2.zip](https://github.com/prusa3d/PrusaSlicer/files/13683683/bridgetester.v2.zip)

Checked a ton of settings in prusaslicer but none of them makes the bridge infill overlap anywhere.
Even went back and tested with vanilla installation and default stock mk3 settings.

Same in prusaslicer 2.7.1, 2.7.0 and older ones (e.g. 2.6.0 tested as well).
superslicer/bambustudio do overlap at least for one perimeter width.

its visible in the slicer preview
[image]
[image]

lukasmatena, 2024-03-21:
Fixed in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1). Closing.

closed 2024-03-21 by lukasmatena