prusa3d/PrusaSlicer #11634: Please add a feature to have open top and/or bottom (non-Spiral vase mode)

laptop-laptop, 2023-11-11:
**Is your feature request related to a problem? Please describe.**

In the Windows 10 version of 2.6.1, I went to Print Settings->Horizontal shells and set the Top and Bottom to zero.
PS stated "Top is open." "Bottom is open." However, PrusaSlicer still printed a very thin layer at the top and bottom
both in the preview and in the actual print.

**Describe the solution you'd like**
Allow the user to have open top and/or bottom without checking Spiral vase.

Jan-Soustruznik, 2023-11-13:
Hello, @laptop-laptop ,

Could you share more info. Thank you

laptop-laptop, 2023-11-22:
I want to print a rectangle without the top and bottom covers. As shown in the screenshots, I went to Print Settings->Horizontal shells and set the Top and Bottom to zero and PS indicated that "Top is open." and "Bottom is open" but in both preview and the actual print, there is a layer on the top and at the bottom.

rtyr, 2023-11-23:
Please provide you 3MF.

closed 2023-11-25 by laptop-laptop