prusa3d/PrusaSlicer #11635: ERROR: Compatible Visual Studio installation not found.

RogerInHawaii, 2023-11-11:
### Description of the bug

I am trying to install the necessary software so that I can look at and make possible modifications to PrusaSlicer.

I'm following the instructions in the "How to build - Windows.md" file for PrusaSlicer.

At one point it says to :

C:\src\PrusaSlicer>git clone https://github.com/prusa3d/PrusaSlicer.git

and it starts off doing it, . . .

Cloning into 'PrusaSlicer'...
remote: Enumerating objects: 224892, done.
remote: Total 224892 (delta 0), reused 0 (delta 0), pack-reused 224892Receiving objects: 100% (224892/224892), 289.56 MiReceiving objects: 100% (224892/224892), 290.75 MiB | 8.06 MiB/s, done.

and so forth.

But at one point is reports:

ERROR: Compatible Visual Studio installation not found.

Previously I had read some instructions that stated that versions of Visual Studio prior to 2019 were incompatible with the requirements of PrusaSlicer, so I went to the Microsoft website and saw that there is an even NEWER version of Visual Studio, namely Microsoft Visual Studio Community 2022, so I installed that instead of the 2019 version.

But now I'm getting that "Compatible Visual Studio installation not found"

So, my question: Is PrusaSlicer ONLY compatible with the 2019 version, NOT with the 2022 version. so it's reporting the error because because of that, and I need to UN-install 2023 and instead install 2019??

OR, is PrusaSlicer actually OK with the 2023 version but the "How to build - Windows.md" file for PrusaSlice is somehow failing to find it on my machine? And if so, how do I get it to find it?

### Project file & How to reproduce

I don't know what this is referring to

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PRUSA SLicer 2.6.1

### Operating system

Windows 10 Home Version 22H2

### Printer model

PRUSA i3 MK3

lukasmatena, 2023-11-11:
It should be possible to use either. I am using 2022 myself. It is difficult to say what the problem is, I would start by using Visual Studio Developer prompt instead of regular cmd.

lukasmatena, 2023-11-11:
Closing, not likely to be PrusaSlicer-related issue.

RogerInHawaii, 2023-11-11:
CLOSING???? This is something that is happening while I'm following the instructions for setting up PrusaSlicer !!!!

It's happening because of a test done within the "How to build - Windows.md" file for PrusaSlicer!

Why is it not finding the Visual Studio installation? It's most definitely installed on my computer? Or if it IS finding it how is it determining that it's NOT COMPATIBLE???

You say that you, yourself, are using 2022. And exactly what "prompt" would I use within Visual Studio Developer?

And it's not that it's reporting "No Visual Studio installation found.". It's reporting "Compatible Visual Studio installation not found." so it seems to be finding the Visual Studio version I have installed, in fact the ONLY Visual Studio version I have installed, which is indeed Visual Studio 2022, but it's deciding that it's not compatible with the current version of PrusaSlicer!!!

The applicable code within the "How to build - Windows.md" file is this:

IF DEFINED PS_VERSION (
    SET /A PS_VERSION_EXCEEDED=%PS_VERSION% + 1
) ELSE SET PS_VERSION=%PS_VERSION_SUPPORTED%
SET MSVC_FILTER=-products Microsoft.VisualStudio.Product.%PS_PRODUCT% -version "[%PS_VERSION%,%PS_VERSION_EXCEEDED%)"
FOR /F "tokens=* USEBACKQ" %%I IN (`^""%VSWHERE%" %MSVC_FILTER% -nologo -property installationPath^"`) DO SET MSVC_DIR=%%I
IF NOT EXIST "%MSVC_DIR%" (
    @ECHO ERROR: Compatible Visual Studio installation not found. 1>&2
    GOTO :HELP
)

So, what is it looking at that it finds is indicating NOT COMPATIBLE ??

lukasmatena, 2023-11-11:
Ok, so only now it is apparent that you are using the `build_win.bat` script. In that case, you can try to use the manual build to see if the problem is in the script or somewhere on your end.

> Why is it not finding the Visual Studio installation?

I really have no idea. The script calls `vswhere`, so either it cannot find it, or `vswhere` cannot find MSVC. If you want further assistance, it might be helpful to actually include the output of the script.

>  And exactly what "prompt" would I use within Visual Studio Developer?

You would open the Developer Command Prompt and run `build_win.bat` from there.

RogerInHawaii, 2023-11-11:
OK, I started clean and did the entire process again, again using the git-cmd.exe for the command window, and following the instructions in the "How to build - Windows.md" file.

In the git-cmd command window I did:

C:\Program Files\Git>cd \

C:\>mkdir src
C:\>cd src
C:\src>git clone https://github.com/prusa3d/PrusaSlicer.git
Cloning into 'PrusaSlicer'...
remote: Enumerating objects: 226326, done.
remote: Counting objects: 100% (1423/1423), done.
remote: Compressing objects: 100% (463/463), done.
remote: Total 226326 (delta 926), reused 1396 (delta 920), pack-reused 224903
Receiving objects: 100% (226326/226326), 302.44 MiB | 8.06 MiB/s, done.

Resolving deltas: 100% (176574/176574), done.
Updating files: 100% (4130/4130), done.

C:\src>cd PrusaSlicer

C:\src\PrusaSlicer>build_win.bat -d=..\PrusaSlicer-deps -r=console

C:\src\PrusaSlicer>REM /|/ Copyright (c) 2022 Jebtrix @Jebtrix

C:\src\PrusaSlicer>REM /|/ Copyright (c) 2021 Justin Schuh @jschuh

C:\src\PrusaSlicer>REM /|/

C:\src\PrusaSlicer>REM /|/ PrusaSlicer is released under the terms of the AGPLv3 or higher

C:\src\PrusaSlicer>REM /|/
**ERROR: Compatible Visual Studio installation not found.**

AND THERE IS IT AGAIN, the ERROR.

The applicable section of the build_win.bat file is:

This is the section for checking whether Visual Studio exists:

REM The officially supported toolchain version is 16 (Visual Studio 2019)
REM TODO: Update versions after Boost gets rolled to 1.78 or later
SET PS_VERSION_SUPPORTED=16
SET PS_VERSION_EXCEEDED=17
SET VSWHERE=%ProgramFiles(x86)%\Microsoft Visual Studio\Installer\vswhere.exe
IF NOT EXIST "%VSWHERE%" SET VSWHERE=%ProgramFiles%\Microsoft Visual Studio\Installer\vswhere.exe
FOR /F "tokens=4 USEBACKQ delims=." %%I IN (`"%VSWHERE%" -nologo -property productId`) DO SET PS_PRODUCT_DEFAULT=%%I
IF "%PS_PRODUCT_DEFAULT%" EQU "" (
    SET EXIT_STATUS=-1
    @ECHO ERROR: No Visual Studio installation found. 1>&2
    GOTO :HELP
)

IT APPARENTLY SUCCEEDS in finding Visual Studio, since it reports no error from the above code.

Later, it does the check for whether or not the found Visual Studio is compatible:

IF DEFINED PS_VERSION (
    SET /A PS_VERSION_EXCEEDED=%PS_VERSION% + 1
) ELSE SET PS_VERSION=%PS_VERSION_SUPPORTED%
SET MSVC_FILTER=-products Microsoft.VisualStudio.Product.%PS_PRODUCT% -version "[%PS_VERSION%,%PS_VERSION_EXCEEDED%)"
FOR /F "tokens=* USEBACKQ" %%I IN (`^""%VSWHERE%" %MSVC_FILTER% -nologo -property installationPath^"`) DO SET MSVC_DIR=%%I
IF NOT EXIST "%MSVC_DIR%" (
    @ECHO ERROR: Compatible Visual Studio installation not found. 1>&2
    GOTO :HELP
)

and it is THAT SECTION that reports "ERROR: Compatible Visual Studio installation not found"

So, it's finding Visual Studio but then determining that it's not compatible (with PrusaSlicer).

So what exactly is wrong? What is it finding that makes it decide that the version of Visual Studio is not compatible, per the above code?

lukasmatena, 2023-11-11:
I looked into it in more detail and you are actually right. It is the `build_win.bat` script who refuses the new MSVC as not worthy, it only supports the 2019 version. When I said that I use the 2022 version, I did not realize that you are calling it through that script. Calling the script with an extra command line parameter `-v 17` should make it work. The script was provided several years ago by @jschuh, at the time when the deps would not build at all using the new version.

We probably should do something about it. My opinion is that since it was provided by the community for the community, it should be the community who maintains it. What we should do is to update the documentation so it explicitly says so.

jschuh, 2023-11-11:
It looks like Boost got rolled forward to a supported version sometime during the development of 2.6. I had a script that periodically checked this, but it seems that it broke at some point.

Anyway, it's a trivial change, so I can post a PR to update things tomorrow.

RogerInHawaii, 2023-11-12:
lukasmatena,

Thank you for looking more closely at this. Are you sure that it's a -v 17 that should be added to the invocation of the build_win.bat batch file? The -v option is indicated as the "Major version number of MSVC installation to use for build". But the instructions for doing the build indicates (somewhere, I don't recall where) that it requires the 2019 version of Visual Studio. I'm guessing that the '17' would be telling it that it needs the 2017 version. PLUS, what I have installed is the 2022 version of Visual Studio. So, should I specify -v 22 ??

jschuh, 2023-11-12:
@RogerInHawaii The product name is "Visual Studio 2022", but the version number is 17. Similarly, "Visual Studio 2019" is version 16. "Visual Studio 6.0" actually had a matching version number of 6 (but then the product name was off by 1992 years 😉). 

More generally, you'll find that projects specify a maximum supported toolchain version because newer versions typically introduce incompatibilities due to language, compiler, or library changes—or even just toolchain bugs. It takes time and effort to update the code to resolve those incompatibilities (e.g. Prusa Slicer didn't build in VS 2022 until earlier this year). That's why it's a good idea to stick to a toolchain version that's explicitly listed as supported.

RogerInHawaii, 2023-11-13:
OK. I started up the batch file, within the Gt CMD window, and included the -v 17 option. It started off pretty good, getting past the point where it previously hit the "cannot find compatible version" error, and it processed for quite some time, displaying updates on what it was doing, but at one point it started reporting warnings, hundreds of warnings, in YELLOW TEXT.

It continued on for a long time and then stopped showing any progress for a long time, although I could see from Task Manager that the Git CMD program was still taking up close to 100%  CPU time.

Finally, after a couple of hours it showed :

**ERROR: *** Build process failed at deps step. ***

C:\src\PrusaSlicer>**

And stopped completely.

**Any suggestions?**

Here's a bit of the output, starting with my invocation of the batch file :

**C:\src\PrusaSlicer>build_win.bat -v 17 -d=..\PrusaSlicer-deps -r=console**

C:\src\PrusaSlicer>REM /|/ Copyright (c) 2022 Jebtrix @Jebtrix

C:\src\PrusaSlicer>REM /|/ Copyright (c) 2021 Justin Schuh @jschuh

C:\src\PrusaSlicer>REM /|/

C:\src\PrusaSlicer>REM /|/ PrusaSlicer is released under the terms of the AGPLv3 or higher

C:\src\PrusaSlicer>REM /|/
**********************************************************************
** Build Config: RelWithDebInfo
** Target Arch:  x64
** Build Steps:  all
** Run App:      console
** Deps path:    C:\src\PrusaSlicer-deps
** Using Microsoft Visual Studio installation found at:
**  C:\Program Files\Microsoft Visual Studio\2022\Community
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.7.6
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
Creating deps\build
Creating C:\src\PrusaSlicer-deps
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
-- The C compiler identification is MSVC 19.37.32825.0
-- The CXX compiler identification is MSVC 19.37.32825.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Forcing CMAKE_BUILD_TYPE to Release as it was not specified.
-- Detected 64-bit compiler => building 64-bit deps bundle
-- Found external package definitions: Blosc;Boost;CGAL;CURL;Catch2;Cereal;EXPAT;GLEW;GMP;JPEG;LibBGCode;MPFR;NLopt;NanoSVG;OCCT;OpenCSG;OpenEXR;OpenSSL;OpenVDB;PNG;Qhull;TBB;TIFF;ZLIB;heatshrink;wxWidgets
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.42.0.windows.2")
-- Mapping dep_ZLIB => dep_Blosc
-- Mapping dep_ZLIB => dep_Boost
-- Mapping dep_Boost => dep_CGAL
-- Mapping dep_GMP => dep_CGAL
-- Mapping dep_MPFR => dep_CGAL
-- Mapping dep_ZLIB => dep_CURL
-- Mapping dep_ZLIB => dep_JPEG
-- Mapping dep_ZLIB => dep_LibBGCode
-- Mapping dep_Boost => dep_LibBGCode
-- Mapping dep_heatshrink => dep_LibBGCode
-- Mapping dep_GMP => dep_MPFR
-- Mapping dep_GLEW => dep_OpenCSG
-- Mapping dep_ZLIB => dep_OpenCSG
-- Mapping dep_ZLIB => dep_OpenEXR
-- Mapping dep_TBB => dep_OpenVDB
-- Mapping dep_Blosc => dep_OpenVDB
-- Mapping dep_OpenEXR => dep_OpenVDB
-- Mapping dep_Boost => dep_OpenVDB
-- Mapping dep_ZLIB => dep_PNG
-- Mapping dep_ZLIB => dep_TIFF
-- Mapping dep_PNG => dep_TIFF
-- Mapping dep_JPEG => dep_TIFF
-- Found OpenGL: opengl32
-- Mapping dep_ZLIB => dep_wxWidgets
-- Mapping dep_PNG => dep_wxWidgets
-- Mapping dep_EXPAT => dep_wxWidgets
-- Mapping dep_TIFF => dep_wxWidgets
-- Mapping dep_JPEG => dep_wxWidgets
-- Mapping dep_NanoSVG => dep_wxWidgets
-- Building dep targets (Release): dep_Boost;dep_ZLIB;dep_CGAL;dep_GMP;dep_MPFR;dep_CURL;dep_Catch2;dep_Cereal;dep_EXPAT;dep_GLEW;dep_LibBGCode;dep_heatshrink;dep_NLopt;dep_OCCT;dep_OpenVDB;dep_TBB;dep_Blosc;dep_OpenEXR;dep_Qhull;dep_wxWidgets;dep_PNG;dep_TIFF;dep_JPEG;dep_NanoSVG
CMake Warning:
  Manually-specified variables were not used by the project:

    DEP_WX_GTK3

-- Building dep targets (Debug): dep_Boost_debug;dep_CGAL_debug;dep_CURL_debug;dep_Catch2_debug;dep_Cereal_debug;dep_EXPAT_debug;dep_GLEW_debug;dep_LibBGCode_debug;dep_NLopt_debug;dep_OCCT_debug;dep_OpenVDB_debug;dep_TBB_debug;dep_Blosc_debug;dep_OpenEXR_debug;dep_Qhull_debug;dep_wxWidgets_debug;dep_PNG_debug
-- Configuring done (16.9s)
-- Generating done (1.0s)
-- Build files have been written to: C:/src/PrusaSlicer/deps/build
MSBuild version 17.7.2+d6990bcfa for .NET Framework

**HERE'S THE FIRST WARNING**
cmake -E tar : warning : skipping symbolic link "libexpat-R_2_4_3/README.md" -> "expat/README.md". [C:\src\PrusaSlicer\
deps\build\dep_EXPAT.vcxproj]
CUSTOMBUILD : CMake warning :  [C:\src\PrusaSlicer\deps\build\dep_Catch2.vcxproj]
CUSTOMBUILD : CMake warning :  [C:\src\PrusaSlicer\deps\build\dep_Cereal.vcxproj]
CUSTOMBUILD : CMake warning :  [C:\src\PrusaSlicer\deps\build\dep_ZLIB.vcxproj]
C:\src\PrusaSlicer\deps\build\dep_ZLIB-prefix\src\dep_ZLIB\gzread.c(319,17): warning C4267: '=': conversion from 'size_
t' to 'unsigned int', possible loss of data [C:\src\PrusaSlicer\deps\build\builds\ZLIB\zlib.vcxproj] [C:\src\PrusaSlice
r\deps\build\dep_ZLIB.vcxproj]

**and it continued on for page after page of similar warnings**

**and finally it showed . . .**

ERROR: *** Build process failed at deps step. ***

C:\src\PrusaSlicer>

jschuh, 2023-11-13:
I can't really say without seeing the actual errors. It's possible you can just do a clean rebuild to fix it. It's also possible you didn't install all of the required Visual Studio components (e.g. if an error says something like "cannot execute" that usually means you're missing a required Visual Studio component).

I do know that the script works fine with a proper install of Visual Studio 2022 or 2019, because I tested clean builds with both yesterday before updating the script.

Really, all of this is pretty typical build environment setup that should be familiar to anyone who's done Windows C++ development. So, I'd suggest working through it on your own, to ensure you're comfortable with the basics. Because setting up the build environment is by far the easy part, compared to understanding and modifying the code.

RogerInHawaii, 2023-11-13:
I installed Visual Studio, accepting all the defaults. Are you saying that I need to know exactly what it is that PrusaSlicer expects so that I can make specific selections for the installation of Visual Studio?

OK, I now notice that one of the final warnings/errors reported during the processing of the build_win.bat batch file indicates a fatal error: INSUFFICIENT DISK SPACE

C:\src\PrusaSlicer\build\src\slic3r\RelWithDebInfo\libslic3r_gui.lib : **fatal error LNK1180: insufficient disk space to
complete link** [C:\src\PrusaSlicer\build\src\slic3r\libslic3r_gui.vcxproj]

I'm attempting to clean up as much as I can on my disk. So far I've got it down to 22.5 Gigabytes of free storage.

Can you tell me how much space is needed in order to install the PrusaSlicer source code for editing, via the build_win.bat batch file??

Also, if it turns that I cannot free up enough space, is it feasible to put it out on my Windows OneDrive?

RogerInHawaii, 2023-11-15:
I tried doing the git clone and the build_win.bat so that the output went out to my Microsoft OneDrive, i.e. out to online storage, instead of directly on to my computer. That didn't work because apparently it (OneDrive) defaults everything to being BOTH online AND on the local computer. So once again it maxed out on the disk space on my local computer.

SO THEN I cleaned up my local C: drive, resulting in 26 Gigabytes of free space and then did the git clone and the build_win.bat so that the output went out to my C: drive. It ran out of space once again.

I realized that I have a D: drive (normally used only for data storage for some reason??) and have over 800 Gigabytes free on it, so I did the git clone and the build_win.bat so that the output went out to that D: drive, and MAYBE it worked.

The git clone part worked fine (as it usually does).

The build_win.bat ran and ran and ran, a couple of hours. 

At one point it reported:

**CMake Warning:
  Manually-specified variables were not used by the project:
  DEP_WX_GTK3**
  
  But it kept going with:
  
  **-- Building dep targets (Debug):**
  
  But then lots of YELLOW warnings started:
  
  **cmake -E tar : warning : skipping symbolic link "libexpat-R_2_4_3/README.md" ->**

and hundreds of warnings like these:

**D:\Prusa Slicer Source Code\src\PrusaSlicer\PrusaSlicer\deps\build\dep_ZLIB-prefix\src\dep_ZLIB\gzread.c(319,17): warni
ng C4267: '=': conversion from 'size_t' to 'unsigned int', possible loss of data**

**warning C4244: 'return': conversion from '__int64' to 'int', possible loss of data**

**warning C4244: 'return': conversion from 'time_t' to 'double', possible loss of data**

It looked like it was finding bad code all over the place, in every source file it processed.

AFTER those hundreds of warnings it reported:

**Cleaning build . . .**

But then showed:

**'Slicer' is not recognized as an internal or external command,
operable program or batch file.**

It then said:

**-- Building for: Visual Studio 17 2022**

And went on to, apparently do a lot of builds.

But then reported:

**Building with Win10 STL fixing service support
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed**

So, something FAILED.

And then:

**-- Looking for pthread_create in pthreads
NOT FOUND ??
-- Looking for pthread_create in pthreads - not found**

So, something NOT FOUND.

It then processed OK (apparently) for a while, but encountered:

**-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)**

More things seemed to go OK for a while.

Then :

**-- Visual Leak Detector (VLD) is not found.**

Then a bunch of processing that seemed to go OK.

It finally got to:

**-- Build files have been written to: D:/Prusa Slicer Source Code/src/PrusaSlicer/PrusaSlicer/build
MSBuild version 17.7.2+d6990bcfa for .NET Framework**

BUT THEN, it started reporting more YELLOW WARNINGS, lots of t hem, like:

**<': signed/unsigned mismatch

The contents of <stdfloat> are available only with C++23 or later.

'==': unsafe mix of type 'char' and type 'const _Ty'**

And then it just stopped. No final message like "Done" or anything, no indication that it was actually completed.

I watched Task Manager the whole time and it did look like the processing time reduced to zero, so I figured it was finally all done.

I looked at the location on the D: drive where it supposedly put everything and, I guess, it looks OK.

Is it now ready for me to use Visual Studio to compile and run and potentially MAKE CHANGES to it?

RogerInHawaii, 2023-11-15:
ONE MORE THING:

When I start up Visual Studio 2022 it has an option to Clone a Repository. Does that mean I could have done all of this, including the build_win.bat part, right from within Visual Studio? Would that build have been executed automatically just by choosing the Clone a Repository option, or would I still have had to initiate that manually, somehow, from within Visual Studio?

Tylersuard, 2024-04-07:
This is a nightmare.  I just downloaded SuperSlicer and got the error at the top, but this conversation has made me decide to not even bother and find a different slicer program.

RogerInHawaii, 2024-04-07:
[Tylersuard](https://github.com/Tylersuard) Do be aware that this is an issue ONLY if your intention is to acquire the actual SOURCE code for PrusaSlicer, usually done when your intention is to make modifications or add to the capabilities of PrusaSlicer (which is what I am doing). It has nothing to do with simply USING  the PrusaSlicer program itself.

closed 2026-09-29 by prusaslicer-bot[bot]