prusa3d/PrusaSlicer #11636: Rotation arrows missing Prusa Slice 2.7.0 Alpha 1 and Beta 1

KaJashey, 2023-11-11:
### Description of the bug

When you want to rotate an object and you place the cursor in the X, Y, or Z rotation you used to get large arrows over an object showing what axis it was going to be rotated on. That doesn't happen in Prusa Slicer 2.7.0 Alpha 1 or Beta 1

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-beta1+MacOS-arm64

### Operating system

MacOS 14.1

### Printer model

MK3S+

JohnOCFII, 2023-11-12:
I can confirm the same behavior (also on macOS - but currently on Ventura 13.6)

Two screen shots below:

[image]

[image]

lukasmatena, 2023-11-12:
Thanks, we will have a look.
SPE-2025

lukasmatena, 2023-11-16:
Fixed in [2.7.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc1). Closing.

closed 2023-11-16 by lukasmatena