prusa3d/PrusaSlicer #11637: Norton 360 deleted files after unzip

michalmpesek, 2023-11-11:
### Description of the bug

Immediately after unzip version 2.7.0beta1, Norton 360 deleted some files. I attached picture with "survied" files.

Version 2.7.0alpha1 working without problems with Norton.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0 beta1

### Operating system

Windows 10 64bit

### Printer model

Prusa MINI+

lukasmatena, 2023-11-11:
This is almost certainly a false positive. I assume that the warning will disappear by itself when the AV becomes aware of it. See https://github.com/prusa3d/PrusaSlicer/issues/6878#issuecomment-914456002 for more info about how it works.

TomasJankauskas5, 2023-11-12:
SAme here norton is blocking everything please fix

lukasmatena, 2023-11-21:
@TomasJankauskas5 Do you have the same problem with 2.7.0-rc1?

TomasJankauskas5, 2023-11-21:
Stoped using it went back to 2.6

/////// www.jnk.design  ////////////////

michalmpesek, 2023-11-21:
> @TomasJankauskas5 Do you have the same problem with 2.7.0-rc1?

Hi, version 2.7.0-rc1 works without problems with Norton 360.

lukasmatena, 2023-11-22:
Thanks, closing.

closed 2023-11-22 by lukasmatena