prusa3d/PrusaSlicer #11639: Print objects not in the center on Prusa XL

nazar-pc, 2023-11-11:
**Is your feature request related to a problem? Please describe.**
When I try to slice something with Prusa Slicer for Prusa XL, it puts objects into the center of the print bed. I suspect there is a way to put objects not in the center, but I was not able to find it quickly in the interface (checked context menu and some of the on-screen buttons).

The issue with that is that print bed on Prusa XL is actually split into 16 segments that heat up independently. By placing a small object in the center, 4 tiles will have to warm up, while if it was moved a bit towards any of the corners, only one tile would need to be heated up, which should reduce power consumption.

**Describe the solution you'd like**
It would be great to make slicer recognize this and move objects such that it covers the least number of heating tiles automatically.

**Describe how it would work**
Object or set of objects placed on a print bed wouldn't be centered by default on printers like Prusa XL. It would take into consideration both X and Y dimensions and would put an object such that i covers the least number of tiles by default. Suggestion to rotate an object to fit one tile might be an interesting option as well, but probably overkill.

**Describe alternatives you've considered**
-

**Additional context**
One of the marketed features of Prusa XL is tiled heating of the printbed, but it is not actually taken advantage of fully most of the time by default.

Jan-Soustruznik, 2023-11-13:
Hello, @nazar-pc , 

I think that feature that you mension above was already implemented into prusaslicer in 2.5.1, when the XL printer was released.

Changelog link to prusaslicer 2.5.1: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.1
> Original Prusa XL specific: Arrangement into the corner
> The Prusa XL printer has a segmented bed, and it is capable of heating just a part of it to lower power consumption. Because the most efficient way of doing this is to start heating at the corner, the Arrange function in PrusaSlicer can now be configured to prefer corners. The setting is available after right-clicking the Arrange icon in the top toolbar and defaults to "Front Left". Auto arrangement is forced when an object is added to an empty print bed.

thank you, have a nice day

nazar-pc, 2023-11-13:
It is close, but not quite what I meant. That option is already set to "Front Left", however when I drop a new object into slicer it appears in dead center and forcefully stays there because "Auto center parts" is enabled in general settings by default and takes precedence.

What I meant bu this feature request is for slicer to be smart enough to put objects into bottom left corner by default, maybe this means "Auto center parts" sholdn't be enabled by default or there should be a prompt to the user about this.

Now that I know what the options are and what they do it makes a lot of sense, but the first time it is not very straightforward.

ex-nerd, 2024-08-30:
See also #11785

closed 2026-09-29 by prusaslicer-bot[bot]