prusa3d/PrusaSlicer #11640: Can't mousewheel scroll through drop-down menus anymore

stereoprologic, 2023-11-11:
### Description of the bug

In 2.6.1, I can quickly mousewheel scroll through infill pattern/percentage and pretty much every other dropdown menu in the settings. In this version, I can't anymore...

### Project file & How to reproduce

See above

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.0-beta1+win64-202311101734_signed.zip 

### Operating system

Windows 10 Pro

### Printer model

Original Prusa MINI

YuSanka, 2023-11-13:
Thanks for your report.
SPE-2026

lukasmatena, 2023-11-16:
Fixed in [2.7.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc1). Closing.

Impact123, 2024-01-05:
While scrolling inside a it now scrolls the window's scrollbar as well. Here's an example where I scroll up and down then simply up at the end. The cursor is clearly over the dropdown.
[image]
```
Version: 2.7.1+win64
Build: PrusaSlicer-2.7.1+win64-202312121425
```

Here's how it used to work on `2.6.1`
[image]

YuSanka, 2024-01-10:
@Impact123 , thanks for your report.
SPE-2102

FidelCapo, 2024-01-29:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2024-01-29 by FidelCapo