prusa3d/PrusaSlicer #11642: File names in gallery view are cutted

yahbluez, 2023-11-12:
### Description of the bug

In the gallery some file names are cutted no matter how big the window is:

[image]

### Project file & How to reproduce

open the gallery selection view

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-beta1

### Operating system

manjaro linux

### Printer model

mk3 and others

YuSanka, 2023-11-13:
Thanks for your report.
SPE-2032

lukasmatena, 2023-11-16:
Fixed in [2.7.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc1). Closing.

closed 2023-11-16 by lukasmatena