prusa3d/PrusaSlicer #11643: PrusaSlicer 2.6.1 crashes when slicing this .STL

skeptopic, 2023-11-12:
### Description of the bug

On Mac, ARM64.

Slicing this file for the Prusa Mini with a 0.6mm nozzle, 0.15mm layer height. Seems to work on other layer heights and nozzles.

Slicing this file crashes immediately 100% of the time, no idea why!

Hope this repro helps to find some nasty little bug!

### Project file & How to reproduce

[BunnyHookv2.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/13327713/BunnyHookv2.stl.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

OSX ARM64

### Printer model

Mini+

lukasmatena, 2023-11-12:
I am unable to reproduce it (on Windows), it would be better if you posted a full 3MF project file.
You can also first try [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). It is possible that it was already fixed.

skeptopic, 2023-11-12:
[repro.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13329122/repro.3mf.zip)

I've inlcuded a repro scene as requested. I have a suspicion you won't find anything, because I also wasn't able to repro on Windows. I strongly suspect this is an ARM64-only bug.

At any rate, I just worked around the problem. I simply submitted to be helpful!

lukasmatena, 2023-11-12:
Ok, we will try on macOS. Can you just confirm that you indeed get the crash with 2.7.0-beta1 too?

JohnOCFII, 2023-11-13:
I was able to reproduce this crash as specified about using PrusaSlicer 2.6.1.  Also on MacBook Pro M1/arm with macOS Sonoma 14.1.1

Here's the first part of the crash log:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [5766]
Path:                  /Applications/Other Apps/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.1+MacOS-x64-202309060812 (PrusaSlicer-2.6.1+MacOS-x64-202309060812)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-11-12 19:35:35.0761 -0600
OS Version:            macOS 14.1.1 (23B81)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       92F68A30-3CAE-4F0A-88A3-DEA922995BCF

Time Awake Since Boot: 26000 seconds
Time Since Wake:       9368 seconds

System Integrity Protection: enabled

Crashed Thread:        13  slic3r_tbb_5

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [5766]

VM Region Info: 0 is not in any region.  Bytes before following region: 4366745600
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104474000-107430000    [ 47.7M] r-x/r-x SM=COW  ...S/PrusaSlicer

Thread 0:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18f7ec8b4 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x18f7fed30 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x18f7f54f0 mach_msg_overwrite + 476
3   libsystem_kernel.dylib        	       0x18f7ecc38 mach_msg + 24
4   CoreFoundation                	       0x18f909ea4 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x18f90876c __CFRunLoopRun + 1208
6   CoreFoundation                	       0x18f907c5c CFRunLoopRunSpecific + 608
7   HIToolbox                     	       0x199e84448 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x199e84284 ReceiveNextEventCommon + 648
9   HIToolbox                     	       0x199e83fdc _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x1930e2c54 _DPSNextEvent + 660
11  AppKit                        	       0x1938b8ebc -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
12  AppKit                        	       0x1930d6100 -[NSApplication run] + 476
13  PrusaSlicer                   	       0x106aa5f64 wxGUIEventLoop::OSXDoRun() + 196
14  PrusaSlicer                   	       0x106b80ba0 wxCFEventLoop::DoRun() + 40
15  PrusaSlicer                   	       0x106b00a6c wxEventLoopBase::Run() + 112
16  PrusaSlicer                   	       0x106adb884 wxAppConsoleBase::MainLoop() + 120
17  PrusaSlicer                   	       0x106a6ba94 wxApp::OnRun() + 36
18  PrusaSlicer                   	       0x106b2e9ec wxEntry(int&, wchar_t**) + 88
19  PrusaSlicer                   	       0x104ebf41c Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 292
20  PrusaSlicer                   	       0x1044831e4 Slic3r::CLI::run(int, char**) + 18084
21  PrusaSlicer                   	       0x10448731c main + 204
22  dyld                          	       0x18f4b10e0 start + 2360
```

JohnOCFII, 2023-11-13:
It did not crash for me with the beta for 2.7.0.

jamesdehart, 2023-11-13:
I happen to get this EXC_BAD_ACCESS (SIGSEGV) on my Mac Book Pro 2017 Intel. The project files are diff but same base error message.

```
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [93868]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.1+MacOS-x64-202309060812 (PrusaSlicer-2.6.1+MacOS-x64-202309060812)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-11-13 09:11:46.9836 -0700
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Bridge OS Version:     3.0 (14Y910)
Anonymous UUID:        07F9D54B-7E66-77C9-4BC6-CF8C5A9EE4A7

Sleep/Wake UUID:       6D934B13-0D39-4A4D-A6A1-3BBAD06900EA

Time Awake Since Boot: 1400000 seconds
Time Since Wake:       1047 seconds

System Integrity Protection: enabled

Crashed Thread:        12  slic3r_tbb_4

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [93868]

VM Region Info: 0 is not in any region.  Bytes before following region: 4357226496
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      103b60000-1071c4000    [ 54.4M] r-x/r-x SM=COW  ...S/PrusaSlicer

Thread 0:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff80589d5a2 mach_msg2_trap + 10
1   libsystem_kernel.dylib        	    0x7ff8058ab71d mach_msg2_internal + 78
2   libsystem_kernel.dylib        	    0x7ff8058a45d4 mach_msg_overwrite + 692
3   libsystem_kernel.dylib        	    0x7ff80589d88a mach_msg + 19
4   CoreFoundation                	    0x7ff8059b800f __CFRunLoopServiceMachPort + 145
5   CoreFoundation                	    0x7ff8059b6a90 __CFRunLoopRun + 1365
6   CoreFoundation                	    0x7ff8059b5ed1 CFRunLoopRunSpecific + 560
7   HIToolbox                     	    0x7ff80f436f3d RunCurrentEventLoopInMode + 292
8   HIToolbox                     	    0x7ff80f436d4e ReceiveNextEventCommon + 657
9   HIToolbox                     	    0x7ff80f436aa8 _BlockUntilNextEventMatchingListInModeWithFilter + 64
10  AppKit                        	    0x7ff808a5225c _DPSNextEvent + 858
11  AppKit                        	    0x7ff808a51106 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1214
12  AppKit                        	    0x7ff808a43788 -[NSApplication run] + 586
13  PrusaSlicer                   	       0x1067b740a wxGUIEventLoop::OSXDoRun() + 186
14  PrusaSlicer                   	       0x1068a269d wxCFEventLoop::DoRun() + 29
15  PrusaSlicer                   	       0x1068190d5 wxEventLoopBase::Run() + 85
16  PrusaSlicer                   	       0x1067f0e24 wxAppConsoleBase::MainLoop() + 116
17  PrusaSlicer                   	       0x10677715a wxApp::OnRun() + 26
18  PrusaSlicer                   	       0x10684a779 wxEntry(int&, wchar_t**) + 73
19  PrusaSlicer                   	       0x10480f078 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 248
20  PrusaSlicer                   	       0x103b6f1b7 Slic3r::CLI::run(int, char**) + 20839
21  PrusaSlicer                   	       0x103b73850 main + 352
22  dyld                          	    0x7ff80558241f start + 1903
```

I would be happy to post all my details if that can help. I will try on Beta slicer.

lukasmatena, 2023-11-13:
If it does not crash in 2.7.0-beta1, it is probably one of the crashes that we already fixed. I cannot tell from the backtraces though, because you only posted the stack of the UI thread, which is not the one that crashed. If you include the crash report in full, we may verify that we indeed fixed it and that it did not disappear by coincidence (quite unlikely in this case).

jamesdehart, 2023-11-24:
@lukasmatena Here is the full report.

[crash_report.txt](https://github.com/prusa3d/PrusaSlicer/files/13454861/crash_report.txt)

lukasmatena, 2023-11-24:
Thanks. Judging by the backtrace, this is definitely a duplicate of #11267, which was fixed in [2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). Closing.

closed 2023-11-24 by lukasmatena