prusa3d/PrusaSlicer #11644: Bridging angle cannot be changed

jonathanmlang, 2023-11-12:
### Description of the bug

Bridging angle setting box has no effect. It always goes whatever direction it wants regardless of what its set to.

### Project file & How to reproduce

[bridging glitch.zip](https://github.com/prusa3d/PrusaSlicer/files/13328330/bridging.glitch.zip)

The bridging angle is wrong at layer 21

Reproduce by opening an stl and setting the bridging angle to anything other than 0

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1+win64

### Operating system

Windows 11

### Printer model

MK3S Prusa

neophyl, 2023-11-12:
Yeah its doesn't, due to that is Bridge Infill.  Bridge Infill is a law unto itself.

Actual Bridging however does obey that angle.  For example here
[image]
the bottom object is an actual bridge (its not over infill) and the direction is controlled through the Bridging angle setting.  

One other thing to note is that the layer height of an actual bridge obeys the Thick Bridges setting too but Bridge Infill also does not.  The bridge infill is in effect the same as thick bridges.  
I too wish there was some actual control over the bridge infill.  I'd like to see it the same layer height (ie thinner) as I've has issues with over extrusion is some small areas that use it.

Jan-Soustruznik, 2023-11-13:
hello @jonathanmlang , 

Bridging angle isn't applied to the infill bridges, like @neophyl mention in his comment above(https://github.com/prusa3d/PrusaSlicer/issues/11644#issuecomment-1807159312) For infill Bridges will be printed using the best direction PrusaSlicer can detect, so this setting does not affect them.

closed 2026-09-29 by prusaslicer-bot[bot]