prusa3d/PrusaSlicer #11645: BGcode

PatB42, 2023-11-12:
### Description of the bug

same problem as the 2.7 alpha 

the prusalink don't accept the Bgcode as you can easy see you can only send or load GCODE files so can't use it with prusalink !!! 
boring

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7 Beta

### Operating system

win 11

### Printer model

mk4

Jan-Soustruznik, 2023-11-13:
Hello, @PatB42, 

thank you for your report. 

If you want to use the new features like .bcode or G2/G3 gcode support you have to install latest FW on your printer(currently in alpha stage) > https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha2 

If you already installed the latest FW on your printer, It may be only string issue(I informed FW team to check it). Uploading .bgcode format should works.

Thank you, have a nice day

PatB42, 2023-11-13:
Thx I use the 5.0.1+12089 today and can’t install the alpha2 via connect i presume it’s normall as it’s alpha one so i’ll install it via usb key and will let u know asap Do you have any idea when the prusalink will be upgrade or repair ? as no control, no sort out by date …missing many option like the link using with the RPiow !?  Thx  Envoyé à partir de Courrier pour Windows De : Jan SoustružníkEnvoyé le :lundi 13 novembre 2023 09:37À : prusa3d/PrusaSlicerCc : PatB42; MentionObjet :Re: [prusa3d/PrusaSlicer] BGcode (Issue #11645) Hello, @PatB42,thank you for your report.If you want to use the new features like .bcode or G2/G3 gcode support you have to install latest FW on your printer(currently in alpha stage) > https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha2If you already installed the latest FW on your printer, It may be only string issue(I informed FW team to check it). Uploading .bgcode format should works.Thank you, have a nice day

PatB42, 2023-11-13:
Hi So far it’s work well with the alpha2 jsut some prolem with the credentials blabla..about the wifi don’t want set up the files on the usb key so i did it with the slicer that great idea and then after manualy it’s work that all but then is goodbye Envoyé à partir de Courrier pour Windows De : Jan SoustružníkEnvoyé le :lundi 13 novembre 2023 09:37À : prusa3d/PrusaSlicerCc : PatB42; MentionObjet :Re: [prusa3d/PrusaSlicer] BGcode (Issue #11645) Hello, @PatB42,thank you for your report.If you want to use the new features like .bcode or G2/G3 gcode support you have to install latest FW on your printer(currently in alpha stage) > https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha2If you already installed the latest FW on your printer, It may be only string issue(I informed FW team to check it). Uploading .bgcode format should works.Thank you, have a nice day

Jan-Soustruznik, 2023-11-13:
Hello, @PatB42, yes thru connect can be install only stable FWs. Current alpha with new features is still in develop stage. I don't know exactly when the final release of new FW will be released, but I think that soon. 

I think that this ticket can be closed as solved. 

Thank you, for your time, have a nice day

closed 2023-11-13 by Jan-Soustruznik

PatB42, 2023-11-13:
yes it's ok with the alpha2 just weird to set up wifi , have to do it via slicer ..but works fine now bgcode accepted by prusalink