prusa3d/PrusaSlicer #11647: Slicer 2.7

TomasJankauskas5, 2023-11-12:
### Description of the bug

Crashes after Any modifier is selected. 

### Project file & How to reproduce

Add part 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7

### Operating system

win10 

### Printer model

Mini+

lukasmatena, 2023-11-12:
Can you please describe the steps in more detail? I can add add and select modifiers all I want. A project/screenshot/screencast might help. Thanks.

TomasJankauskas5, 2023-11-13:
I think this is part where Norton antivirus really doesn't like this alpha
software. I will let you know and we'll update more

/////// www.jnk.design  ////////////////

lukasmatena, 2023-11-21:
@TomasJankauskas5 Any update? Do you have any issues with 2.7.0-rc1? Thanks.

closed 2026-09-29 by prusaslicer-bot[bot]