prusa3d/PrusaSlicer #11648: Dynamic perimeter fan speeds are being applied to infill

jschuh, 2023-11-12:
### Description of the bug

It looks like the fan speed is not getting reset when an external perimeter completes, so the last dynamic fan speed ends up getting applied to all of the infill. In the picture below the infill should be dark blue (no fan), but you can see that it's instead using the fan speed from the perimeter seam.

[image]

This showed up in the first 2.7 alpha, and continues to be present in the most recent beta.

### Project file & How to reproduce

This is a simple test file that I used to generate the image above: [external_perimeter_fan_test.zip](https://github.com/prusa3d/PrusaSlicer/files/13328968/external_perimeter_fan_test.zip)

It just uses an extreme set of of fan speeds to make it easier to visualize what's happening.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0-beta1

### Operating system

Windows 11 (but it's an OS independent issue)

### Printer model

Prusa MK3S (but it's a printer independent issue)

jschuh, 2023-11-12:
I nerd-sniped myself into taking a look. The problem is here: https://github.com/prusa3d/PrusaSlicer/blob/87a58670f3a20dbda2478e35f7b74663d36c099b/src/libslic3r/GCode.cpp#L3267-L3271

Since a fan reset always happens at the end of an extrusion, the resulting string looks like ";_EXTRUDE_END;_RESET_FAN_SPEED". So in `CoolingBuffer::parse_layer_gcode` it always hits this path: https://github.com/prusa3d/PrusaSlicer/blob/87a58670f3a20dbda2478e35f7b74663d36c099b/src/libslic3r/GCode/CoolingBuffer.cpp#L506 
And thus this later conditional is just unreachable code:
https://github.com/prusa3d/PrusaSlicer/blob/87a58670f3a20dbda2478e35f7b74663d36c099b/src/libslic3r/GCode/CoolingBuffer.cpp#L571

Given that the cooling buffer code already uses bit flags to manage this state, it sure seems like the set/reset flags need only be conditioned on each other, and not the rest of the flag state. In my testing that change produces correct output, so I'll post a PR.

jschuh, 2023-11-12:
I just posted PR #11650. This is what it looks like with the PR applied, which is also how the same file looks in version 2.6.

[image]

jschuh, 2023-11-13:
I poked around a bit more and found that the regression was introduced in fa0986c0c5548282a955644d40b5f170604dc913. Previously the [";_RESET_FAN_SPEED" was always wrapped in newlines](https://github.com/prusa3d/PrusaSlicer/commit/fa0986c0c5548282a955644d40b5f170604dc913#diff-cb3d09d2073b6e9032c65d07aa8f10d166324a53136b2aa1288f7b76f7d94319L2997). So, one potential fix is to just add the newlines back, but that seems more fragile to me than updating the flag setting in CoolingBuffer.cpp. Although, I tested and verified that either approach works.

hejllukas, 2023-11-14:
I merged your PR (with the modification that I mentioned there) into our private repository, so the commit will appear with the next release.
Thank you for detailed investigation of the issue and proposed ways to fix it.

lukasmatena, 2023-11-16:
Fixed in [2.7.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc1). Closing.
Thanks to @jschuh for the fix provided in #11650).

closed 2023-11-16 by lukasmatena