prusa3d/PrusaSlicer #11649: The physical printer dialog is very difficult to use after the 2.7 changes

neclimdul, 2023-11-12:
### Description of the bug

The physical printer selection has always been pretty awkward and the 2.7 changes look very nice but they've made it almost impossible to actually select the correct profile for my printer.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0-beta1+linux-x64-GTK3

### Operating system

Ubuntu 23.04 x86_64

### Printer model

Ender 5 Pro

kocikdav, 2023-11-13:
Hello, thank you. You mean the right side panel dropdown menu with printers right?  Unfortunately we are currently limited how to show long names. But there should be tooltip with full name available. Is it visible for you?

lukasmatena, 2023-11-13:
I would also like to ask about the difference between 2.6.1 and 2.7.0-beta1. As far as I know, the contents of the dropdown did not change, has it?

neclimdul, 2023-11-14:
I don't know that this is the right solution for everyone but for me the printer model isn't really useful information and I'd prefer if it went away. Couple reasons why.

 1. You have to duplicate the nozzle information in the physical printer name to keep it unique.
 2. I generally know the printer type based on the physical printer name.
 3. I know the printer type for the name so its not useful
 4. If I _want_ the printer type i can put it in the physical name, but shorter.

Alternatively, maybe just putting the profile first would be better since that's the primary purpose of the dialog?

Either way its not useful having 5 characters to describe your profile.

neclimdul, 2023-11-24:
Sadly in the full 2.7.0 release, profiles are still unusable.  And unfortunately it looks like I haven't responded because of the label but there' nothing I can do :(

neclimdul, 2026-09-28:
I see the new alpha has a new profile system, but last I checked it didn't support using existing 3rd party profiles without some scripts I couldn't get working. Happy to test 3.0 with 3rd party profiles again once that's working to see if its still difficult to use.