prusa3d/PrusaSlicer #11652: Embossing limitations - Multimaterial options for embossed text and wrapping issues.

tommylux, 2023-11-13:
**Describe the solution you'd like**
When using the text tool to create a cut, it would be nice if you could then use the multimaterial painter to select the cut elements and colour code as necessary. However, multimaterial painter doesn't recognise the new shape (even if you tried to export the whole model as STL and re-import.

The only way to colour the text is to selet it as a modifier and then not have the ability to emboss the text.

**Is your feature request related to a problem? Please describe.**
Not able to set colour for cut out elemnents using the text tool.

**Describe how it would work**
Perhaps an extruder option could appear on the right hand objects pane to indicate the inside colour. The text modifier in the object pane doesn't have a dropdown list for extruder. Or, once you have added the text element, you coudl then choose the main body and use multi material print to be able to smart select the cut out faces. If easier, perhaps you could 'flatten' the modification first, then treat it as a new STL.

**Describe alternatives you've considered**
Best alternitive is to not use the feature and just use a modifier instead. Or use 3D Builder in windows to handle the embossing.

**Additional context**

GuyH77, 2023-11-13:
I'd love to see this too.

murk-sy, 2023-11-16:
As a workaround you can add modifiers to an object, and then export and re-import it. Painting it, however, can be a chore.

[image]

Would be nice if this was doable in-slicer too (and ideally reversible) since dealing with more complex part arrangements can get difficult.

tommylux, 2023-11-16:
I thought I treid exporting and importing, I didn't recall the shape was percistent with the modifier. I'd have to try it again.
But yes, It would be fantastic if it is included in the slicer.

murk-sy, 2023-11-16:
You have to specifically select the parent object, otherwise it exports just the unmodified part (or the modifier, whichever is selected).

Jony01, 2023-11-22:
When you would like to modify bottom of cutted text, You could copy/paste (ctrl+c & ctrl+v) negative volume.
Than set bigger depth(to be over bottom of cutted text) plus set it as modifier with different color(extruder).
[image]

tommylux, 2023-11-22:
Thanks for the tips, appreciated.

[image]<https://user-images.githubusercontent.com/15888762/284857421-55cb1ef1-7c3a-4364-a86e-3e58459682de.png>

tommylux, 2024-11-04:
Still hopeful of this feature being released.. Can't quite get the modifier workaround to work.

Jony01, 2024-11-05:
As I mentioned before. 
1. Selected text (e.g. Have opened "_Emboss_" window)
2. [ctrl + c] than [ctrl + v]
In right panel(object list) will appear new volume
NOTE1: You should be in Slicer [image]
NOTE2: New copy&pasted text overlap original one.
3. Change text to be "_Modifier_".
4. Increase "_Depth_" to be deeper than cutted part
5. Change color of modifier(e.g. in right panel)
[image]

tommylux, 2024-11-05:
> As I mentioned before.
>
> 1. Selected text (e.g. Have opened "_Emboss_" window)
>
> 2. [ctrl + c] than [ctrl + v]
>
> In right panel(object list) will appear new volume
>
> NOTE1: You should be in Slicer [...]

Thanks, got it to work, it worked well. I had to set depth of modifier (much) further than the negative part, to cover a perimeter width. Thanks for the workaround suggestion. [image]

closed 2026-09-29 by prusaslicer-bot[bot]