prusa3d/PrusaSlicer #11653: Text wrapping limitations when using the Text tool

tommylux, 2023-11-13:
**Is your feature request related to a problem? Please describe.**
The text wrapping around an object is vastly restricted, often I cannot add text to an obect such as model dog because the angles are too high. In comparrison, you can achive the desired by using microsoft's 3D Builder quite easily (however, this app is not available on Mac).

**Describe the solution you'd like**
To be able to support more angles, to support the text cut out at x mm into the object regardless of the angle.

Jan-Soustruznik, 2023-11-13:
Hello, @tommylux , could you share a model, please? Thank you

thomasluxton, 2023-11-13:
Hi, Thanks for getting back to me. I've taken another look and it doesn't seem bad as I was making out, my apologies.
[image]
[image]

I think what I was noticing what the placement of the text element seems to jump around quite a lot as there is texture on the model. Here is a screen grab while placing:
[image]

After a short while, the preview is rendered:
[image]

Perhaps the texture of the object could be simplified when placing text to stop it jumping around so much.

The experience seems to be easier in 3D Builder:
[image]
[image]

tommylux, 2023-11-13:
Hi, Thanks for getting back to me. I've taken another look and it doesn't seem bad as I was making out, my apologies.
[image]
[image]

I think what I was noticing what the placement of the text element seems to jump around quite a lot as there is texture on the model. Here is a screen grab while placing:
[image]

After a short while, the preview is rendered:
[image]

Perhaps the texture of the object could be simplified when placing text to stop it jumping around so much.

The experience seems to be easier in 3D Builder:
[image]
[image]

closed 2026-09-29 by prusaslicer-bot[bot]