prusa3d/PrusaSlicer #11656: Automatic placement and inactive models

Aldaris89, 2023-11-13:
### Description of the bug

Automatic placement of models doesn't work correctly when there are inactive elements, because of the automatic placement, all models end up outside the workspace. 

In one of the older versions of PrusaSlicer (I dont remember witch one), this worked correctly. Setting an element as inactive and running automatic placement would place all inactive elements outside the table, while active elements were placed on the table surface.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-beta1

### Operating system

Win10

### Printer model

MK3S+

neophyl, 2023-11-13:
Check if you have Auto Centre parts enabled in your preferences.  That will do this.  As the alphas/betas/release versions all store to different places its possible to have settings in preferences that are different between the 3 types.

Aldaris89, 2023-11-13:
Sorry, I didn't know this option was for that. Thank you!

closed 2023-11-13 by Aldaris89