prusa3d/PrusaSlicer #11657: Report problem button not working

Dexter0007, 2023-11-13:
### Description of the bug

When I clicked button "Report problem" nothing happened.

### Project file & How to reproduce

[image]

PrusaSlicer
Version:   2.6.1+win64
Build:     PrusaSlicer-2.6.1+win64-202309060711

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (budowa 19045), wydanie 64-bitowe
Total RAM size [MB]: 8,504MB
OpenGL installation
GL version:   4.6.0
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1050/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1+win64-202309060711

### Operating system

Windows 10 (budowa 19045), wydanie 64-bitowe

### Printer model

MK3S+

neophyl, 2023-11-13:
Just tried it on mine, same version and same OS.  It pops up a dialog 
[image]

Given that on Yes it just opens a new tab/page and No just closes the dialog and that there's a tick box for remember choice is it possible you have previously selected no and remembered choice ?

lukasmatena, 2023-11-13:
It is very likely to be caused by what @neophyl has said. If that's the case, you can change it in `Preferences->Other`. Does this solve the issue?

Dexter0007, 2023-11-13:
I don't remember the "open hyperlink" dialogue. Perhaps another user on my computer had this window.
In preferences I have settings like this below
[image]

Dexter0007, 2023-11-13:
Problem was in the Brave browser. Reinstallation the browser resolve the issue

closed 2023-11-13 by Dexter0007