prusa3d/PrusaSlicer #11658: Terrible Travel Optimization

dondec123, 2023-11-13:
### Description of the bug

Instead of traveling to "nearest neighbor" for next extrusions, PS is traveling all the way to the other side, back and forth .  

This is both Horizontally and Vertically for this multi color print, adding hours to this job.  Example Video Here: 
https://1drv.ms/v/s!Arf1YeMUEh5Gm8FOvR0XbR4_BdF3jw?e=OgWm0a 

I cut 2 hrs off this 17hr print by simply turning off Avoid Crossing Perimeters.   But it still travels all the way across the print, back and forth for the extrusions.

Something's wrong with the Travel Optimizer.  Please check.  Thank you.

### Project file & How to reproduce

Project File Here:
https://1drv.ms/u/s!Arf1YeMUEh5Gm8FMm5D7Bb_BLRiR8w?e=oFbRGE 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0 beta1

### Operating system

Win 11

### Printer model

Formbot Troodon2.0 (voron2.4 clone) w AfterBurner.

1701Pietko, 2024-03-23:
Yeah. Prusa Slicer 2.7.2 and the same things happens to me. The nozzle travels illogically from side to side of the print... when it should finish one side first and then to go to other side. 
It happens also for support. I'm sure if the travel path was better, significant time would be spared.

unrepentantgeek, 2024-05-03:
I've experienced this myself.  It appears to be a regression of https://github.com/prusa3d/PrusaSlicer/issues/5511

Here is a test case part I made up to show the problem.
[Path.planner.test.case.zip](https://github.com/prusa3d/PrusaSlicer/files/15205344/Path.planner.test.case.zip)

You can clearly see the same shape getting entirely different path plans depending on whether it's single or multi-material.

Experienced this on my XL.

ex-nerd, 2024-05-03:
I see this a lot when printing multi-color (MMU painted) versions of my honeycomb wall storage bins (https://www.printables.com/model/804857-parametric-hsw-hex-shelf-bin). I just now discovered that rotating some of these parts 90° can cut a huge chunk off of the print time (if this bug didn't exist, it should not matter, and I suspect fixing the bug would improve things even further).

At their request, I emailed a 3mf project to Prusa several months ago but haven't heard back.

ex-nerd, 2024-05-03:
Oh, and this is definitely still happening in `2.7.4`

parhelia, 2024-05-19:
Please fix this bug as it seems to be a regression.

https://github.com/prusa3d/PrusaSlicer/assets/429148/62ab5947-263a-4c7c-a76b-8ce315f77b33

Same model file sliced with Bambu Studio 1.9.1.67 doesn't have this problem.

https://github.com/prusa3d/PrusaSlicer/assets/429148/c7188786-d1bb-4f1c-a710-d267945fbed0

ex-nerd, 2024-06-26:
There was a comment in the earlier 2.8 release notes that suggested an improvement might have been related to this, however, I just checked `2.8rc2` with the test file from @unrepentantgeek and this is still happening:

[image]

blastrock, 2024-10-13:
This is still an issue. I have this design with a lot of holes, and these holes are made in a seemingly random order. This piece takes 1h51m of just travel time!

[image]

The project file: [lid.zip](https://github.com/user-attachments/files/17356662/lid.zip)

I've skimmed through the code and saw comments that suggest that this optimization should have been made: https://github.com/prusa3d/PrusaSlicer/blob/5dc04b4e8f14f65bbcc5377d62cad3e86c2aea36/src/libslic3r/Arachne/PerimeterOrder.cpp#L247-L248 and this is from 17th of July : https://github.com/prusa3d/PrusaSlicer/commit/86309ba939c17ae6e35f0cecb5db6f3032c71fe8

dondec123, 2024-10-13:
I'd like to know why this has not been fixed?!!?
This is a serious Travel Optimization bug.

    Don

image.png (view on web)<https://github.com/user-attachments/assets/6594c028-9782-4a09-8216-46f50e6482a6>

I've skimmed through the code and saw comments that suggest that this optimization should have been made: https://github.com/prusa3d/PrusaSlicer/blob/5dc04b4e8f14f65bbcc5377d62cad3e86c2aea36/src/libslic3r/Arachne/PerimeterOrder.cpp#L247-L248 and this is from 17th of July : 86309ba<https://github.com/prusa3d/PrusaSlicer/commit/86309ba939c17ae6e35f0cecb5db6f3032c71fe8>

philiip11, 2024-10-20:
I'm trying to print this very detailed model. Due to the terrible travel optimization, even the tiniest bit of stringing adds up to mess of spiderwebs. Over 25% of the print time are travel moves. I'd really appreciate it if someone could fix this :-)

[image]
[image]

1701Pietko, 2024-10-20:
> I'm trying to print this very detailed model. Due to the terrible travel optimization, even the tiniest bit of stringing adds up to mess of spiderwebs. Over 25% of the print time are travel moves. I'd really appreciate it if someone could fix this :-) <

I don't think that in your case this travel time is avoidable. it is a feature of printing the multicolored object.

juckendesAuge, 2024-10-31:
Can you test whether the Travel Moves are drastically reduced if you set the Perimeter Generator from Arachne to Classic?
This is the case for me, but there is an error with the Honneycomb (not infill).
See picture

Classic Perimeter Generator:
[image]

philiip11, 2024-10-31:
> I don't think that in your case this travel time is avoidable.

Sure it is. It would save a lot of time if those bricks would be printed in a clockwise or counterclockwise order instead of jumping from left to right and back all the time.

> Can you test whether the Travel Moves are drastically reduced if you set the Perimeter Generator from Arachne to Classic? 

For me it changes from 5h45m (Arachne) to 5h51m (Classic).

juckendesAuge, 2024-10-31:
@philiip11 
The total printing time will probably be more for classic, as the perimeter width is always the same.
However, the travel movements are reduced in my case.
See video

https://github.com/user-attachments/assets/a16828b5-5e4f-42d2-bd7e-2363aa83b383

SmallGreenParr0ot, 2024-11-21:
I am having the same issue on a single colour model, printed in PETG.
https://www.printables.com/model/1082331-sunlu-s2-circulation-air-duct 
The 3mf file can be accessed at Printables.
"Avoid Crossing Perimeters" is turned on to minimize stringing.
The slicing of the model creates a number if islands of similar print feature types and there is a lot of non-printing travel between these islands. 
A typical example can be seen at Z=10.4
It appears this is due to the gcode being generated by print feature type, creating many long non-print travel paths.

[image]

SmallGreenParr0ot, 2024-12-03:
Those two issues seem to be more broad, including travel issues related to external perimeters, where seam settings will affect the travel.
This was intended to focus on travel related to how the slicer handles the different types of fill.
For example, on that layer, the slicer completes all the fill of one type in one arm and then moves to the next arm and does the same.
In this case, it would be travel optimal to complete the two fill types in one arm, then move to the next arm and complete the two fill types, etc.
What seems to be missing in the slicer is an extra pass at the end to remove the gcode sorting by fill type.

SmallGreenParr0ot, 2024-12-03:
The example which initiated this post has many islands and the lack of travel optimization between islands.
My example has a single island and the lack if travel optimization is noticeable when traveling between fill types in the single island.
But the issue is the same; the underlying sorting generated by the slicer has not been removed, both within in a single island, and the islands themselves.
What this points to is a missing travel optimization pass which ignores embedded gcode sorting.

SmallGreenParr0ot, 2024-12-04:
A note on the relative importance of this issue: Yes, it’s a pain to wait longer for a print.
But E3D and others have invested in developing expensive, high flow nozzles and Prusa has been investing in research into increasing the print speed. Why?
Of course it’s to reduce print time.
But because the slicer doesn’t have optimized travel, that, in my case, increased the overall print time in the order of 15-20%.
Is the print time reduction that could be achieved by optimizing travel such a complex issue that it’s not being addressed?
The answer to that can be seen in the performance of other slicers.
I’m left to wonder why this issue is marked as low priority.

ex-nerd, 2024-12-04:
@SmallGreenParr0ot It's not just about speed, either. All of those retractions and back/forth often leave strings and other surface quality issues in both single- and multi-color prints that affect both appearance and part strength.

SmallGreenParr0ot, 2024-12-05:
I’m not sure where this request should go.
From my own experience, the absence of travel optimization is not always a big issue, depending on the model being sliced and printed.
Hopefully, when a remedy is made, it will be in the form of an additional user-selectable action in PrusaSlicer.
I had in mind an additional “Optimize Now” button near the “Slice Now” button.
This would allow the user to keep the fast slicing time on models where travel optimization is not a big issue.
And it would also allow the user to accept spending more time to optimize the previously generated slice output, when print time is much more important than slicing time.

dondec123, 2024-12-06:
The problem is absolutely REAL for those doing Multi Color prints.

A separate and fully optional OPTIMIZE function after slicing makes sense to me.  Perhaps it could appear when Multiple Extruders are enabled.

Time lost to "dumb and meaningless" travel moves... Is Significant.

I hope resources are assigned to fix this issue soon.  We've suffered far too long.

AndKe, 2024-12-08:
The quality of printing big honeycomb stuctures is terrible.  Oozing during the meaningless travels makes the results ugly.

SmallGreenParr0ot, 2024-12-09:
This problem exists not only for multiple extruders. The sample 3mf file I provided that demonstrates the issue is only one colour, one extruder. So the optimize facility needs to be available for all projects.

AndKe, 2024-12-09:
+1 on that, I follow this issue because the total mess it makes it affecting single-color prints as well.

this :
[image]

is printed like this:
[image]

- which is a horror-show quality-wise.

EightBitAlex, 2024-12-10:
I'm currently printing large plates of parts that have a U shape and it really shows this travel inefficient travel move pathing. I made up a crude model to demonstrate the issue. In this simple case anything where a single part has at least 2 islands to print the slicer will finish the paths related to that part first instead of hoping to the nearest island. On my actual part plates this results in 20% of the print time being travel moves, which for large batch plates it really adds up quickly. 

I haven't been able to figure out a setting to optimize this, but some of my old slicers handle this particular aspect of travel moves much more logically. 
[image]
[Test-Part.zip](https://github.com/user-attachments/files/18083870/Test-Part.zip)

In this example, ideally there should be 1 move across the large gap instead 5

AndKe, 2024-12-10:
This is not much that says "failure" as the slicing result of this part:

[slice-this.zip](https://github.com/user-attachments/files/18084083/slice-this.zip)

SmallGreenParr0ot, 2024-12-10:
I am disappointed in the response to this posted issue.

There should be no judgement of “pass” or “fail”, it should be taken in with improvement in mind.
In fact, this issue should be at the top of the rather large pile.

So here is an open invitation to Josef Průša to personally review the issue.

Surely adding 20% travel time wipes away all the efforts and investment in MK4 print speed improvement? 
  
The question I have is:-

“Why listen to the claims of the Prusa MK4 print speed improvement when PrusaSlicer generates 20% travel times?”

AndKe, 2024-12-11:
Good point, everyone like to brag about speed and quality. this mess sabotages BOTH speed and quality.

SmallGreenParr0ot, 2024-12-11:
There is little joy in posting more details on this issue.
It seems there is little or no interest in fixing it.
The appearance of SuperSlicer, forked from PrusaSlicer, is telling.
https://github.com/supermerill/SuperSlicer
I have sliced both my example project and the rather larger project which initiated this issue, with SuperSlicer and the output is much better.  
So for anyone who is experiencing this issue, my suggestion is to try SuperSlicer.
The settings are different so there is a small learning curve.
But as it was forked from PrusaSlicer, it has the same look and feel.
So my thanks go to SuperSlicer.

mgurzixo, 2024-12-11:
My contribution:

With arachne:
[image]

Total time: 1h2m
Travel time: 15m (24.8%)

With old perimeter generator:
[image]

Total time: 54m
Travel time: 7m (12.5%)

Just scraping arachne.

I attach the STL as a test case.

Michel

[Grille.zip](https://github.com/user-attachments/files/18103012/Grille.zip)

philiip11, 2025-02-18:
Fixed with PrusaSlicer-2.9.1-alpha1

closed 2025-02-18 by lukasmatena

SmallGreenParr0ot, 2025-04-11:
I have not viewed the slicer code but it's clear there is significant nesting in producing the travel commands. So the only solution is to add another pass over all the gcode to optimize it.
Don't hold your breath for this to be "fixed".  
This is not a bug in the slicer. It is a feature  that PrusaSlicer does not have.
New features are appearing but this is not a simple one. So it's either wait patiently or use another slicer.

dzwiedziu-nkg, 2026-09-04:
Maybe my plugin for Prusa Slicer 3.0 will help: https://forum.prusa3d.com/forum/prusaslicer/thank-you-for-the-plugin-direction-in-3-0-three-slicing-plugins-it-made-possible-and-a-pr/