prusa3d/PrusaSlicer #11659: Two support types

jldevoy, 2023-11-13:
Hi,
Would it be possible to add the ability to use 2 different support types in a print? I need a strong support to hold the mesh but also a light one that easily breakable for internal moving parts.

Thanks

rtyr, 2023-11-14:
You can use different support types for individual objects. It can be set in object list. Combining more support types for one object is not supported nor planned.

Tupson444, 2023-11-14:
It's understandable if it's difficult to implement, but it would be really good to have, because organic supports are way batter for some areas while standard supports are still the best option for other (e.g. large flat floating surface, where organic would require many small branches, which is slower and would either have many Z hops and thus slow the print even more, or have a risk of collisions and causing print fail).

closed 2023-11-14 by lukasmatena