prusa3d/PrusaSlicer #11661: Height range Modifier - Perimeters

Rhoenbussard, 2023-11-13:
### Description of the bug

When I change the number of perimeters in the Height range Modifier, I get an ugly object. In the range of change it is greater than the former layers.
I tested it with a cube and find same differences
[Shape Box Results.pdf](https://github.com/prusa3d/PrusaSlicer/files/13340187/Shape.Box.Results.pdf)

Please help me

### Project file & How to reproduce

[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/13340266/Shape-Box.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Sovol SV01

rtyr, 2023-11-14:
I was not able to reproduce your g-code output with your project and PS version.

Box2
[image]

Box5
[image]

Rhoenbussard, 2023-11-14:
I can't reproduce the settings, when I try to create another example. Today all was fine.
Sorry

rtyr, 2023-11-23:
Ok, I am closing the issue. We can reopen if it occurs again.

closed 2023-11-23 by rtyr