prusa3d/PrusaSlicer #11664: 2.7.0 Beta 1: GUI Checkboxes is now very hard to see in GTK2 version

Area5142, 2023-11-14:
### Description of the bug

All GUI check boxes is changed and is now nearly impossible to see if checked or not.

Example check boxes in 2.6.1:
[image]

Example check boxes in 2.7.0 Beta 1:
[image]

In both examples the check boxes has the same values.
Looking closely the "new" check boxes has a tiny dot when selected.

### Project file & How to reproduce

Just open PS and view any dialog with a check box.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0 Beta 1

### Operating system

Linux Mint 21.1 Cinnemon

### Printer model

All

Area5142, 2023-11-14:
In PS 2.7.0 Beta 1 GTK3 version, the check boxes are invisible when not checked:
[image]

OS: Linux Mint 21.1 Cinnemon, dark mode

lukasmatena, 2023-11-16:
Fixed in [2.7.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-rc1). Closing.

closed 2023-11-16 by lukasmatena