prusa3d/PrusaSlicer #11666: Post processing script warning everytime I open a project.

Taromisaki2020, 2023-11-14:
### Description of the bug

Post processing script warning everytime I open a project. Please add the option to disable the notification that the 3mf file contains a post process. Because I use one to generate the thumbnails in all my projects and it is very annoying every time I open a file that the message appears and I have to click ok to continue.Thank you so much!

[image]

### Project file & How to reproduce

[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/13344875/Shape-Box.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0 Beta 1

### Operating system

Windows 10

### Printer model

ender 3 V2

carlosjln, 2024-01-19:
This alert is SO annoying 😩 please add a checkbox "do not show this again" 🙏

Jan-Soustruznik, 2024-02-06:
Hello, @Taromisaki2020 ,

Thank you for your report. We created internal task SPE-2138 in our tracker. We will try to make a improvements in future, but now this isn't our priority task.

Have a nice day

jkotowicz123, 2025-02-12:
+1 Need an option in settings to disable it - I use my own post-processing script all the time, so it asks me every single time I open PrusaSlicer...

closed 2026-09-29 by prusaslicer-bot[bot]